hoeken/PsychicHttp #97: Crash on event.send() after a while.

hitecSmartHome, 2024-06-03:
Hi! Amazing lib.
I had a crash on event.send() the last day.

```cpp
==================== CURRENT THREAD STACK =====================
#0  0x401d1464 in httpd_sess_get (hd=0x40000de8, sockfd=1073743892) at C:\Users\Pc\.platformio\packages\framework-espidf@3.40407.0\components\esp_http_server\src\httpd_sess.c:167
#1  0x401d1ddc in httpd_socket_send (hd=0x40000de8, sockfd=1073743892, buf=0x3fff2434 <error: Cannot access memory at address 0x3fff2434>, buf_len=557, flags=0) at C:\Users\Pc\.platformio\packages\framework-espidf@3.40407.0\components\esp_http_server\src\httpd_txrx.c:605
#2  0x401ad5f1 in PsychicEventSourceClient::sendEvent (this=0x3ffe4364, event=0x3fff2434 <error: Cannot access memory at address 0x3fff2434>) at lib/PsychicHttp/src/PsychicEventSource.cpp:151
#3  0x401ad792 in PsychicEventSource::send (this=0x3ffb3c58 <events>, message=<optimized out>, event=<optimized out>, id=1717333565, reconnect=0) at lib/PsychicHttp/src/PsychicEventSource.cpp:123
#4  0x40174017 in HsHServer::sendEvent (this=<optimized out>, type=0x3f409767 "ThermState", doc=...) at src/Server/Event.cpp:60
#5  0x40125f99 in <lambda()>::operator() (__closure=<optimized out>) at src/Components/Thermostat/Thermostat.cpp:156
#6  std::_Function_handler<void(), Thermostat::sendPeriodicData()::<lambda()> >::_M_invoke(const std::_Any_data &) (__functor=...) at c:\users\pc\.platformio\packages       oolchain-xtensa-esp32@8.4.0+2021r2-patch5\xtensa-esp32-elf\include\c++\8.4.0\bits/std_function.h:297
#7  0x400d4135 in std::function<void ()>::operator()() const (this=0x3ffd1000) at c:\users\pc\.platformio\packages      oolchain-xtensa-esp32@8.4.0+2021r2-patch5\xtensa-esp32-elf\include\c++\8.4.0\bits/std_function.h:687
#8  0x40193027 in Sys::handleIntervals (this=0x3ffb3eac <hshSystem>) at src/System/SystemInterval.cpp:55
#9  0x4018c9e1 in Sys::loop (this=0x3ffb3eac <hshSystem>) at src/System/System.cpp:545
#10 0x4018ca08 in systemTask (pvParameters=<optimized out>) at src/System/System.cpp:554
```

I have this wrapper around `events.send()`

```cpp
void HsHServer::sendEvent(const char* type, JsonDocument& doc) {
    if (!hasEventClient() && !serverRouter.connected()) { return; }
    const size_t size = measureJson(doc) + 1;
    char* buffer;
    if (!hshSystem.alloc(buffer, size)) { return; }
    serializeJson(doc, buffer, size);
    events.send(buffer, type, hshTime.getEpoch());
    serverRouter.redirect(buffer,type);
    free(buffer);
}
```

This method called frequently.
The crash clearly was in here

```cpp
#0  0x401d1464 in httpd_sess_get (hd=0x40000de8, sockfd=1073743892) at C:\Users\Pc\.platformio\packages\framework-espidf@3.40407.0\components\esp_http_server\src\httpd_sess.c:167
#1  0x401d1ddc in httpd_socket_send (hd=0x40000de8, sockfd=1073743892, buf=0x3fff2434 <error: Cannot access memory at address 0x3fff2434>, buf_len=557, flags=0) at C:\Users\Pc\.platformio\packages\framework-espidf@3.40407.0\components\esp_http_server\src\httpd_txrx.c:605
#2  0x401ad5f1 in PsychicEventSourceClient::sendEvent (this=0x3ffe4364, event=0x3fff2434 <error: Cannot access memory at address 0x3fff2434>) at lib/PsychicHttp/src/PsychicEventSource.cpp:151
#3  0x401ad792 in PsychicEventSource::send (this=0x3ffb3c58 <events>, message=<optimized out>, event=<optimized out>, id=1717333565, reconnect=0) at lib/PsychicHttp/src/PsychicEventSource.cpp:123
```

The esp was working great. It was up for 2 and a half days but crashed with this error.

Any idea?

hitecSmartHome, 2024-06-03:
This portion of the code looks unfinished in `PsychicEventSource.cpp`

```cpp
void PsychicEventSourceClient::sendEvent(const char *event) {
  if( !this->server() || !this->socket() ){
    return;
  }
  int result;
  do {
    result = httpd_socket_send(this->server(), this->socket(), event, strlen(event), 0);
  } while (result == HTTPD_SOCK_ERR_TIMEOUT);

  //if (result < 0)
  //error log here
}
```

`httpd_socket_send` API states the following

```
Returns:

Bytes : The number of bytes sent successfully - HTTPD_SOCK_ERR_INVALID : Invalid arguments - HTTPD_SOCK_ERR_TIMEOUT : Timeout/interrupted while calling socket send() - HTTPD_SOCK_ERR_FAIL : Unrecoverable error while calling socket send()
Note:
This API is not recommended to be used in any request handler. Use this only for advanced use cases, wherein some asynchronous data is to be sent over a socket. This internally calls the default send function, or the function registered by httpd_sess_set_send_override().
```

Why is there a do while loop? Can't we just do this?

```cpp
void PsychicEventSourceClient::sendEvent(const char *event) {
  if( !this->server() || !this->socket() ){return;}
  httpd_socket_send(this->server(), this->socket(), event, strlen(event), 0);
}
```

or better

```cpp
bool PsychicEventSourceClient::sendEvent(const char *event) {
  if( !this->server() || !this->socket() ){return;}
  int result = httpd_socket_send(this->server(), this->socket(), event, strlen(event), 0);
  if( result == HTTPD_SOCK_ERR_INVALID || result == HTTPD_SOCK_ERR_TIMEOUT || result == HTTPD_SOCK_ERR_FAIL ){
      return false;
  }
  return true;
}
```

hitecSmartHome, 2024-06-03:
For sure something is fishy in here

hoeken, 2024-08-09:
Working through old issues here.  Have you tried the suggestions?

The do() loop was so that it could attempt to resend if there was a timeout error.  It might be that the client disconnected, it looped around, and then failed.  I think moving the '''if( !this->server() || !this->socket() ){return;}''' code into the do() loop would fix this.

hitecSmartHome, 2024-08-11:
Will refresh the lib on my project tomorrow and have a check. It still crashes for me when two or more users are using the web.

hoeken, 2024-08-11:
@hitecSmartHome i just pushed a pretty major rewrite to v2-dev which will hopefully fix this.  commit 376dfb22292137adabaaa6f80bb97466621964a1

hitecSmartHome, 2024-08-11:
Cool. Will check!

mhaberler, 2024-08-11:
@hitecSmartHome you might want to try this: https://github.com/hoeken/PsychicHttp/pull/158

I had it only crash once and unfortunately lost the backtrace, but `PsychicEventSource::send` was the location

hitecSmartHome, 2024-08-12:
Hm... I will patch this commit only and will see.

hoeken, 2024-08-12:
@hitecSmartHome v2-dev branch ran overnight for me, sending 40msg/sec to 2 clients.   no problems so far.

if you could please test against this branch and report back, i'd really appreciate it. :)

mhaberler, 2024-08-12:
the trigger to the crash I saw were appropriately timed client disconnects from an EventSource 

so steady state runs might not trigger it

hitecSmartHome, 2024-08-12:
I have implemented the commit only from @mhaberler suggestion. The crash was definitely triggered by two or more clients connection and disconnection. I'm testing with one of my tester guy. We will give it a couple of days and I will report back

hitecSmartHome, 2024-08-18:
It seems stable with #158 so far.

hitecSmartHome, 2024-08-21:
Okay, it crashes.

```cpp
PC: 0x401ca4f7
EXCVADDR: 0x0002542c

Decoding stack results
0x401ca4f4: httpd_sess_get at C:\Users\Pc\.platformio\packages\framework-espidf@3.40407.0\components\esp_http_server\src\httpd_sess.c line 167
0x401cae09: httpd_socket_send at C:\Users\Pc\.platformio\packages\framework-espidf@3.40407.0\components\esp_http_server\src\httpd_txrx.c line 605
0x401a8f52: PsychicEventSourceClient::sendEvent(char const*) at lib/PsychicHttp/src/PsychicEventSource.cpp line 151
0x401a90db: PsychicEventSource::send(char const*, char const*, unsigned int, unsigned int) at C:/Users/Pc/.platformio/packages/framework-arduinoespressif32/cores/esp32/WString.h line 339
0x401756ad: HsHServer::sendEvent(char const*, ArduinoJson::V704PB22::JsonDocument&) at src/Server/Event.cpp line 6
```

hoeken, 2024-08-21:
> Decoding stack results0x401ca4f4: httpd_sess_get at ***@***.***\components\esp_http_server\src\httpd_sess.c line 1670x401cae09: httpd_socket_send at ***@***.***\components\esp_http_server\src\httpd_txrx.c line 6050x401a8f52: PsychicEventSourceClient::sendEvent(char const*) at lib/PsychicHttp/src/PsychicEventSource.cpp line 1510x401a90db: PsychicEventSource::send(char const*, char const*, unsigned int, unsigned int) at C:/Users/Pc/.platformio/packages/framework-arduinoespressif32/cores/esp32/WString.h line 3390x401756ad: HsHServer::sendEvent(char const*, ArduinoJson::V704PB22::JsonDocument&) at src/Server/Event.cpp line 6

Is this with 1.2.1 or v2-dev?

mhaberler, 2024-08-21:
from the line numberiing I'd guess v2-dev

hitecSmartHome, 2024-08-21:
I have only applied the patch

hitecSmartHome, 2024-08-21:
```cpp
void PsychicEventSource::send(const char *message, const char *event, uint32_t id, uint32_t reconnect){
  if( !message || !event ){ return; }
  String ev = generateEventMessage(message, event, id, reconnect);
  for(PsychicClient *c : _clients) {
    if (c && c->_friend) {
      ((PsychicEventSourceClient*)c->_friend)->sendEvent(ev.c_str());
    }
  }
}
```

mhaberler, 2024-08-21:
is there a semaphore for _clients? 
maybe there should be - sounds like a disconnect and _friend evaporated during the loop

hitecSmartHome, 2024-08-21:
I'm sure it is that. It happens when multiple clients are connected to the same event handler.
The app is sending events rapidly and the users sometimes pushing the app into the background which sometimes causes the http socket to close and when the client click to the app again, it reconnects. It can happen in less than a second sometimes.

mhaberler, 2024-08-21:
give it a try, define a semaphore and do this around all locations modifying _client or _client._friend

example:  https://github.com/theelims/ESP32-sveltekit/commit/b5898d665f2f24901b36ac9254af407c1ccbef76

hitecSmartHome, 2024-08-22:
I think this `xSemaphoreTake(clientSubscriptionsMutex, portMAX_DELAY);` is not a good approach. It will wait `portMAX_DELAY` if the semaphore is taken. It is a really long time and can froze a task. If I add these semaphores into the library, I have to create my own buffering in a separate task.

hitecSmartHome, 2024-08-22:
Imagine if a task calls `events.send()` and it does this

```cpp
void PsychicEventSource::send(const char *message, const char *event, uint32_t id, uint32_t reconnect){
  if( !message || !event ){ return; }
  String ev = generateEventMessage(message, event, id, reconnect);
  for(PsychicClient *c : _clients) {
    xSemaphoreTake(eventClientMutex, portMAX_DELAY);
    if (c && c->_friend) {
      ((PsychicEventSourceClient*)c->_friend)->sendEvent(ev.c_str());
    }
    xSemaphoreGive(eventClientMutex);
  }
}
```

This will froze the caller task for `portMAX_DELAY` time if the semaphore is taken elsewhere.

hitecSmartHome, 2024-08-22:
This would not be a problem If I had an other task which only handles the sending of the events.
If I create a wrapper around `events.send()` which buffers the events and a separate task sends them, it would be ok. But it will still froze the sender task for `portMAX_DELAY` and the events will not be realtime

hitecSmartHome, 2024-08-22:
Crashed again with one connected client.

```cpp
==================== CURRENT THREAD STACK =====================
#0  0x401ca580 in httpd_sess_get (hd=0x40000de8, sockfd=1073743892) at C:\Users\Pc\.platformio\packages\framework-espidf@3.40407.0\components\esp_http_server\src\httpd_sess.c:167
#1  0x401cae90 in httpd_socket_send (hd=0x40000de8, sockfd=1073743892, buf=0x3f838524 <error: Cannot access memory at address 0x3f838524>, buf_len=302, flags=0) at C:\Users\Pc\.platformio\packages\framework-espidf@3.40407.0\components\esp_http_server\src\httpd_txrx.c:605
#2  0x401a8fd9 in PsychicEventSourceClient::sendEvent (this=0x3ffe0454, event=0x3f838524 <error: Cannot access memory at address 0x3f838524>) at lib/PsychicHttp/src/PsychicEventSource.cpp:151
#3  0x401a9162 in PsychicEventSource::send (this=0x3ffb4628 <events>, message=<optimized out>, event=<optimized out>, id=1724267698, reconnect=0) at lib/PsychicHttp/src/PsychicEventSource.cpp:122
#4  0x401756e3 in HsHServer::sendEvent (this=0x3ffb453c <hshServer>, type=0x3f402a79 "HeaterState", doc=...) at src/Server/Event.cpp:60
#5  0x400efb38 in Heater::sendSync (this=0x3f871ecc) at src/Components/Heater/Utils.cpp:53
#6  0x400ebc7c in <lambda(int, int)>::operator() (__closure=0x3ffcb914, prev=<optimized out>, now=<optimized out>) at src/Components/Heater/Heater.cpp:11
#7  std::_Function_handler<void(int, int), Heater::Heater(ArduinoJson::V704PB22::JsonObject&)::<lambda(int, int)> >::_M_invoke(const std::_Any_data &, int &&, int &&) (__functor=..., __args#0=@0x3ffcb8e0: 57, __args#1=@0x3ffcb8e4: 58) at c:\users\pc\.platformio\packages oolchain-xtensa-esp32@8.4.0+2021r2-patch5\xtensa-esp32-elf\include\c++\8.4.0\bits/std_function.h:297
#8  0x40197e99 in std::function<void (int, int)>::operator()(int, int) const (this=0x3ffcb914, __args#0=<optimized out>, __args#1=<optimized out>) at c:\users\pc\.platformio\packages oolchain-xtensa-esp32@8.4.0+2021r2-patch5\xtensa-esp32-elf\include\c++\8.4.0\bits/std_function.h:687
#9  0x40197fac in Time::emitSecChange (now=58, prev=57, this=<optimized out>) at src/Time/TimeEvents.cpp:69
#10 Time::emitSecChange (this=<optimized out>, prev=57, now=58) at src/Time/TimeEvents.cpp:62
#11 0x40197fcc in Time::monitorSecChange (this=0x3ffb4af4 <hshTime>) at src/Time/TimeEvents.cpp:331
#12 0x40198130 in Time::handleEvents (this=0x3ffb4af4 <hshTime>) at src/Time/TimeEvents.cpp:369
#13 0x40196cd5 in Time::loop (this=0x3ffb4af4 <hshTime>) at src/Time/Time.cpp:220
#14 0x4019fe57 in loop () at src/main.cpp:95
#15 0x401c05dc in loopTask (pvParameters=<optimized out>) at C:/Users/Pc/.platformio/packages/framework-arduinoespressif32/cores/esp32/main.cpp:50
```

hitecSmartHome, 2024-08-26:
Okay so I have created a separate task just for sending the events. When a task calls this wrapper function, it will push it into a buffer. I have also placed semaphores around my events buffer. Still got a crash but it is a little bit different this time.

```cpp
==================== CURRENT THREAD STACK =====================
#0  0x401a95f9 in PsychicEventSource::send (this=0x3ffb4630 <events>, message=<optimized out>, event=<optimized out>, id=1724583763, reconnect=0) at lib/PsychicHttp/src/PsychicEventSource.cpp:120
#1  0x401756c0 in HsHServer::consumeEvents (this=<optimized out>) at src/Server/Event.cpp:28
#2  HsHServer::consumeEvents (this=0x3ffb453c <hshServer>) at src/Server/Event.cpp:16
#3  0x40175701 in HsHServer::EventTask (pvParameters=0x0) at src/Server/Event.cpp:11

======================== THREADS INFO =========================
  Id   Target Id          Frame
* 1    process 1073579320 0x401a95f9 in PsychicEventSource::send (this=0x3ffb4630 <events>, message=<optimized out>, event=<optimized out>, id=1724583763, reconnect=0) at lib/PsychicHttp/src/PsychicEventSource.cpp:120
  2    process 1073574320 0x3ffd6f90 in ?? ()
  3    process 1073610188 Shader::getCurrent (this=0x3f86ff54) at src/Components/Shader/Utils.cpp:224
  4    process 1073527396 0x401c02f4 in serialEventRun () at C:/Users/Pc/.platformio/packages/framework-arduinoespressif32/cores/esp32/HardwareSerial.cpp:65
  5    process 1073601508 esp_timer_impl_get_counter_reg () at C:\Users\Pc\.platformio\packages\framework-espidf@3.40407.0\components\esp_timer\src\esp_timer_impl_lac.c:128
  6    process 1073469252 ArduinoJson::V704PB22::detail::parseNumber (s=0x3ffce30c "180", result=...) at lib/ArduinoJson-7.x/src/ArduinoJson/Numbers/parseNumber.hpp:64
  7    process 1073594108 0x4008ca12 in vTaskDelay (xTicksToDelay=1) at C:\Users\Pc\.platformio\packages\framework-espidf@3.40407.0\components\freertos asks.c:1601
  8    process 1073473676 0x40266e4a in esp_pm_impl_waiti () at C:\Users\Pc\.platformio\packages\framework-espidf@3.40407.0\components\esp_pm\pm_impl.c:855
  9    process 1073475576 0x40266e4a in esp_pm_impl_waiti () at C:\Users\Pc\.platformio\packages\framework-espidf@3.40407.0\components\esp_pm\pm_impl.c:855
  10   process 1073609580 0x4008ca12 in vTaskDelay (xTicksToDelay=2) at C:\Users\Pc\.platformio\packages\framework-espidf@3.40407.0\components\freertos asks.c:1601
  11   process 1073538728 0x4008b22a in xQueueReceive (xQueue=0x3f824384, pvBuffer=0x3ffcfb40, xTicksToWait=<optimized out>) at C:\Users\Pc\.platformio\packages\framework-espidf@3.40407.0\components\freertos\queue.c:1487
  12   process 1073413408 0x4008b374 in xQueueSemaphoreTake (xQueue=0x3ffaf6bc, xTicksToWait=<optimized out>) at C:\Users\Pc\.platformio\packages\framework-espidf@3.40407.0\components\freertos\queue.c:1687
  13   process 1073411220 0x4008b374 in xQueueSemaphoreTake (xQueue=0x3ffaee30, xTicksToWait=<optimized out>) at C:\Users\Pc\.platformio\packages\framework-espidf@3.40407.0\components\freertos\queue.c:1687
  14   process 1073607284 0x4008b22a in xQueueReceive (xQueue=0x3f82f8b4, pvBuffer=0x3ffc37e8, xTicksToWait=<optimized out>) at C:\Users\Pc\.platformio\packages\framework-espidf@3.40407.0\components\freertos\queue.c:1487
  15   process 1073467392 0x4000bff0 in ?? ()
  16   process 1073606800 0x40082d70 in esp_crosscore_int_send_yield (core_id=1) at C:\Users\Pc\.platformio\packages\framework-espidf@3.40407.0\components\esp_system\crosscore_int.c:145
  17   process 1073552972 0x4000bff0 in ?? ()
  18   process 1073477988 0x4008d5ac in prvProcessTimerOrBlockTask (xListWasEmpty=<optimized out>, xNextExpireTime=0) at C:\Users\Pc\.platformio\packages\framework-espidf@3.40407.0\components\hal\esp32\include/hal/cpu_ll.h:39
  19   process 1073471564 0x4008b22a in xQueueReceive (xQueue=0x3f823b4c, pvBuffer=0x3ffd0410, xTicksToWait=<optimized out>) at C:\Users\Pc\.platformio\packages\framework-espidf@3.40407.0\components\freertos\queue.c:1487
  20   process 1073550560 0x4008b22a in xQueueReceive (xQueue=0x3f823a6c, pvBuffer=0x3ffd1440, xTicksToWait=<optimized out>) at C:\Users\Pc\.platformio\packages\framework-espidf@3.40407.0\components\freertos\queue.c:1487
```

The log points me to this function

```cpp
void PsychicEventSource::send(const char *message, const char *event, uint32_t id, uint32_t reconnect){
  if( !message || !event ){ return; }
  String ev = generateEventMessage(message, event, id, reconnect);
  for(PsychicClient *c : _clients) {
    if (c && c->_friend) {
      ((PsychicEventSourceClient*)c->_friend)->sendEvent(ev.c_str());
    }
  }
}
```

This is the function which sends the events

```cpp
void HsHServer::consumeEvents(){
    if( eventMessageIndex == 0 ){ return; }

    if( xSemaphoreTake(eventMutex, EVENT_SEM_WAIT) != pdTRUE ){
       return;
    }

    for (int i = 0; i < eventMessageIndex; i++) {
        EventMessage& message = eventMessages[i];
        events.send(message.message, message.type, message.epoch);
        free(message.message);
        free(message.type);
    }
    eventMessageIndex = 0;
    xSemaphoreGive(eventMutex);
}
```

Now I will place semaphores insde the lib where `_client` is used.

mhaberler, 2024-08-26:
the criticial resource is the clients and _friends respectively - AFAICT there is no chance that more than one thread would SEND events

what about taking out the guesswork for us and publishing an example repo to reproduce

the crystal ball unfortunately is hazy today

hitecSmartHome, 2024-08-26:
ofc there is a chance. I have multiple threads on multiple cores. Each can call events.send() at the same time. But I have proved to myself that the problem is not in my part, because it crashes even when a separate thread manages the sending with guards around it.

> what about taking out the guesswork for us and publishing an example repo to reproduce

unfortunately the problem is not that easily reproducible. I certainly can't trigger it that easily. Will put semaphores around _client and we will see.

hitecSmartHome, 2024-08-26:
I started to place semaphores where any `PsychicClient` is used but these are all over the place. I have created a `SemaphoreGuard.h` which will help me managing semaphores.

```cpp
#pragma once

#include <Arduino.h>

class SemaphoreGuard {
public:
    // Constructor that takes a SemaphoreHandle_t by reference
    explicit SemaphoreGuard(SemaphoreHandle_t& mutex) : _mutex(mutex), _acquired(false) {
        if (_mutex != nullptr) {
            _acquired = xSemaphoreTake(_mutex, portMAX_DELAY) == pdTRUE;
        }
    }

    // Destructor that releases the semaphore if it was successfully acquired
    ~SemaphoreGuard() {
        if (_acquired) {
            xSemaphoreGive(_mutex);
        }
    }

    // Deleted copy constructor and assignment operator to avoid accidental copies
    SemaphoreGuard(const SemaphoreGuard&) = delete;
    SemaphoreGuard& operator=(const SemaphoreGuard&) = delete;

private:
    SemaphoreHandle_t& _mutex;
    bool _acquired;
};
```

This way I only have to drop this `SemaphoreGuard guard(clientMutex);` into the start of a scope and will handle the semaphore destruction for me. Example:

```cpp
// Base class
class PsychicHandler {
  protected:
  SemaphoreHandle_t clientMutex;
}
// Constructor
PsychicHandler::PsychicHandler() :
  _filter(NULL),
  _server(NULL),
  _username(""),
  _password(""),
  _method(DIGEST_AUTH),
  _realm(""),
  _authFailMsg("")
  {
    clientMutex = xSemaphoreCreateMutex();
  }

// Method
PsychicClient * PsychicHandler::checkForNewClient(PsychicClient *client)
{
  SemaphoreGuard guard(clientMutex);
  PsychicClient *c = PsychicHandler::getClient(client);
  if (c == NULL)
  {
    c = client;
    addClient(c);
    c->isNew = true;
  }
  else
    c->isNew = false;

  return c;
}
```

I have created `clientMutex` in `PsychicHandler` but `PsychicHttpServer` has no idea about `PsychicHandler` so the compiler comlpains about the use of `clientMutex` 😠

hitecSmartHome, 2024-08-26:
I don't understand why don't you guys have any problem with this.

hoeken, 2024-08-26:
@hitecSmartHome It might be because you are the only one using this in production or stress testing it like this.  We'll get it sorted though.

Can you confirm if you are using the v2 EventSource code?  In particular the new code that uses the 'proper' httpd_queue_work call?  https://github.com/hoeken/PsychicHttp/blob/92ee3e266803ce1fbf9464ddc71c7e9f40a38ec5/src/PsychicEventSource.cpp#L156

Previously we were abusing httpd_socket_send() and I have a feeling that is what is causing your crashes.

As for reproduction, maybe you could ask ChatGPT to create a demo page that opens and closes eventsource connections at random to trigger a crash?

hitecSmartHome, 2024-08-27:
I dont use the v2. There is my own tweaks, thats why the line numbering did not match. Will upgrade to v2 today and release a beta package. I did try to build a reproduction sketch but could not trigger it. It is probably triggered some Iphone specific handling with the event source. I wouldnt be surprised at least. I think the problem comes when the user do not close the web app but place it in the background. Alter an undefined time, Iphone freezes the connections. Both ws and sse. But when you get the app back into focus, my code checks when it was last unfocused. And if it is more than 30 sec, we force both the sse and ws to reconnect and resync the data. I think that is the point where something happens. Maybe there will be two sse connections on the same client or something. But first I will try the v2.

hitecSmartHome, 2024-08-27:
So should I download the `v2-dev` branch?

hitecSmartHome, 2024-08-27:
Okay, I have started to migrate to the v2 branch. I had a lot of problems but now the sketch compiles but it does not work. I get an `Unable to allocate memory.` response when I try to load a page from littlefs. It should be the default page when I type the IP in the URL.

hitecSmartHome, 2024-08-27:
```cpp
E (35482) psychic: Unable to allocate 0 bytes to send chunk
```
interesting

hitecSmartHome, 2024-08-27:
Well. I will do that. It does not serve any file from my file system.

hoeken, 2024-08-27:
Yeah sorry we committed a bunch of changes but need to test it all still.
That will happen soon once I get back from vacation

hitecSmartHome, 2024-08-27:
I think I can't easily copy and paste the changes from PsychicEventSource.
There are a lot of things incompatible and undefined.

hoeken, 2024-08-28:
Its really just the event.send() code.  You should be able to replace the sendEvent() function and add the extra callbacks and it will work:

Add to top of event source header:

```cpp
typedef std::function<void(PsychicEventSourceClient* client)> PsychicEventSourceClientCallback;

typedef struct {
    httpd_handle_t handle;
    int socket;
    char* event;
    size_t len;
    transfer_complete_cb callback;
    void* arg;
} async_event_transfer_t;
```

Add to PsychicEventSourceClient in header:
```cpp
    esp_err_t _sendEventAsync(httpd_handle_t handle, int socket, const char* event, size_t len);
    static void _sendEventWorkCallback(void* arg);
    static void _sendEventSentCallback(esp_err_t err, int socket, void* arg);
```

Replace/add to EventSource.cpp:
```cpp
void PsychicEventSourceClient::sendEvent(const char* event)
{
  _sendEventAsync(this->server(), this->socket(), event, strlen(event));
}

esp_err_t PsychicEventSourceClient::_sendEventAsync(httpd_handle_t handle, int socket, const char* event, size_t len)
{
  // create the transfer object
  async_event_transfer_t* transfer = (async_event_transfer_t*)calloc(1, sizeof(async_event_transfer_t));
  if (transfer == NULL) {
    return ESP_ERR_NO_MEM;
  }

  // populate it
  transfer->arg = this;
  transfer->callback = _sendEventSentCallback;
  transfer->handle = handle;
  transfer->socket = socket;
  transfer->len = len;

  // allocate for event text
  transfer->event = (char*)malloc(len);
  if (transfer->event == NULL) {
    free(transfer);
    return ESP_ERR_NO_MEM;
  }

  // copy over the event data
  memcpy(transfer->event, event, len);

  // queue it.
  esp_err_t err = httpd_queue_work(handle, _sendEventWorkCallback, transfer);

  // cleanup
  if (err) {
    free(transfer->event);
    free(transfer);
    return err;
  }

  return ESP_OK;
}

void PsychicEventSourceClient::_sendEventWorkCallback(void* arg)
{
  async_event_transfer_t* trans = (async_event_transfer_t*)arg;

  // omg the error is overloaded with the number of bytes sent!
  esp_err_t err = httpd_socket_send(trans->handle, trans->socket, trans->event, trans->len, 0);
  if (err == trans->len)
    err = ESP_OK;

  if (trans->callback)
    trans->callback(err, trans->socket, trans->arg);

  // free our memory
  free(trans->event);
  free(trans);
}

void PsychicEventSourceClient::_sendEventSentCallback(esp_err_t err, int socket, void* arg)
{
  // PsychicEventSourceClient* client = (PsychicEventSourceClient*)arg;

  if (err == ESP_OK)
    return;
  else if (err == ESP_FAIL)
    ESP_LOGE(PH_TAG, "EventSource: send - socket error (#%d)", socket);
  else if (err == ESP_ERR_INVALID_STATE)
    ESP_LOGE(PH_TAG, "EventSource: Handshake was already done beforehand (#%d)", socket);
  else if (err == ESP_ERR_INVALID_ARG)
    ESP_LOGE(PH_TAG, "EventSource: Argument is invalid (#%d)", socket);
  else if (err == HTTPD_SOCK_ERR_TIMEOUT)
    ESP_LOGE(PH_TAG, "EventSource: Socket timeout (#%d)", socket);
  else if (err == HTTPD_SOCK_ERR_INVALID)
    ESP_LOGE(PH_TAG, "EventSource: Invalid socket (#%d)", socket);
  else if (err == HTTPD_SOCK_ERR_FAIL)
    ESP_LOGE(PH_TAG, "EventSource: Socket fail (#%d)", socket);
  else
    ESP_LOGE(PH_TAG, "EventSource: %#06x %s (#%d)", (int)err, esp_err_to_name(err), socket);
}
```

Or just wait a couple weeks and we'll get all the bugs worked out of v2-dev.

mathieucarbou, 2024-08-28:
> Okay, I have started to migrate to the v2 branch. I had a lot of problems but now the sketch compiles but it does not work. I get an `Unable to allocate memory.` response when I try to load a page from littlefs. It should be the default page when I type the IP in the URL.

What is the size of the page ? More than 8k ? Chunk mode is activated if the content is more than: `FILE_CHUNK_SIZE` which is set to 8k by default (imo this is too much). This could be overridden with a compile flags: -DFILE_CHUNK_SIZE=2048.

Serving files concurrently is also not a great idea since it causes a lot of interruptions and will multiply the heap buffers required to read them and/or send the chunks.

These ESP boards are too limited in terms of memory and CPU to correctly support concurrent requests. At least this is my opinion.

I would rather have a code focused on serving fast one request at a time (which is what Arduino Webserver is doing and also ESPAsyncWebServer to an extent) instead of trying to support multiple requests at the same time which can also lead to other memory access issues on user code besides.

All the required temporary buffers to be able to either enqueue or send data are impacting heap consumption and fragmentation.

hitecSmartHome, 2024-08-28:
since it is using malloc for allocating the chunks, and I have set malloc to allocate from psram if the allocation is above 64 byte, it should not matter. There is plenty of ram available from my external ram which is 4mb. The initial page size is 4.1kb zipped btw but i have 164 kb files too. My web app is built with webpack which is chunking it so there are a lot of small files scattered. The size should not matter. Also there wasn't any problem before. My frontend requests a ton of files on load and it is fast A.F and reliable. I think the ESP is more than enough for these tasks.

> Its really just the event.send() code. You should be able to replace the sendEvent() function and add the extra callbacks and it will work:

Will try it, thank you very much!

mathieucarbou, 2024-08-28:
@hitecSmartHome : the only way to get `unable to allocate memory` errors is when malloc cannot allocate.

See: https://github.com/search?q=repo%3Ahoeken%2FPsychicHttp%20%22Unable%20to%20allocate%20memory%22&type=code

Maybe you could try to add some prints of the heap/psram before these allocation requests to assert there is enough memory like you suspect?

It's possible that in v2 there is a leak somewhere since we didn't have time to verify all the ne code yet, but if there is one, at least you should see it through there measurements...

hitecSmartHome, 2024-08-28:
Will check it for sure, but yes, it is not related. I have applied the new SSE changes from the V2 as Hoeken told me. We will see the results. I also reverted my buffering and just directly sending the events as it was before, but with the lib changes.

BCsabaEngine, 2024-09-01:
> Will check it for sure, but yes, it is not related. I have applied the new SSE changes from the V2 as Hoeken told me. We will see the results. I also reverted my buffering and just directly sending the events as it was before, but with the lib changes.

In order to help hitecSmartHome's work, we already manage the v2 version of psychicHttp: https://github.com/BCsabaEngine/svelteesp32/discussions/27

hitecSmartHome, 2024-09-09:
It crashes with the new `SSE` implementation too.

```cpp
==================== CURRENT THREAD STACK =====================
#0  httpd_queue_work (handle=0x40000de8, work=0x401a8124 <PsychicEventSourceClient::_sendEventWorkCallback(void*)>, arg=0x3ffe0eb0) at C:\Users\Pc\.platformio\packages\framework-espidf@3.40407.0\components\esp_http_server\src\httpd_main.c:87
#1  0x401a8361 in PsychicEventSourceClient::_sendEventAsync (this=<optimized out>, handle=0x40000de8, socket=1073743892, event=0x3ffeaa20 <error: Cannot access memory at address 0x3ffeaa20>, len=108) at lib/PsychicHttp/src/PsychicEventSource.cpp:179
#2  0x401a83a8 in PsychicEventSourceClient::sendEvent (this=0x3ffe0454, event=0x3ffeaa20 <error: Cannot access memory at address 0x3ffeaa20>) at lib/PsychicHttp/src/PsychicEventSource.cpp:150
#3  0x401a852e in PsychicEventSource::send (this=0x3ffb4628 <events>, message=<optimized out>, event=<optimized out>, id=1725798746, reconnect=0) at lib/PsychicHttp/src/PsychicEventSource.cpp:125
#4  0x401743e0 in HsHServer::sendEvent (this=<optimized out>, type=0x3f41e3a3 "time", doc=...) at src/Server/Event.cpp:71
#5  0x4017a5f1 in <lambda(int, int)>::operator() (__closure=0x3ffcb9e4, prevSec=<optimized out>, nowSec=<optimized out>) at src/Server/Server.cpp:121
#6  std::_Function_handler<void(int, int), HsHServer::handleClientTimeSync()::<lambda(int, int)> >::_M_invoke(const std::_Any_data &, int &&, int &&) (__functor=..., __args#0=<optimized out>, __args#1=<optimized out>) at c:\users\pc\.platformio\packages oolchain-xtensa-esp32@8.4.0+2021r2-patch5\xtensa-esp32-elf\include\c++\8.4.0\bits/std_function.h:297
#7  0x40197179 in std::function<void (int, int)>::operator()(int, int) const (this=0x3ffcb9e4, __args#0=<optimized out>, __args#1=<optimized out>) at c:\users\pc\.platformio\packages        oolchain-xtensa-esp32@8.4.0+2021r2-patch5\xtensa-esp32-elf\include\c++\8.4.0\bits/std_function.h:687
```

hitecSmartHome, 2024-09-09:
```cpp
==================== THREAD 1 (TCB: 0x3ffcbb2c, name: 'loopTask') =====================
#0  httpd_queue_work (handle=0x40000de8, work=0x401a8124 <PsychicEventSourceClient::_sendEventWorkCallback(void*)>, arg=0x3ffe0eb0) at C:\Users\Pc\.platformio\packages\framework-espidf@3.40407.0\components\esp_http_server\src\httpd_main.c:87
#1  0x401a8361 in PsychicEventSourceClient::_sendEventAsync (this=<optimized out>, handle=0x40000de8, socket=1073743892, event=0x3ffeaa20 <error: Cannot access memory at address 0x3ffeaa20>, len=108) at lib/PsychicHttp/src/PsychicEventSource.cpp:179
#2  0x401a83a8 in PsychicEventSourceClient::sendEvent (this=0x3ffe0454, event=0x3ffeaa20 <error: Cannot access memory at address 0x3ffeaa20>) at lib/PsychicHttp/src/PsychicEventSource.cpp:150
#3  0x401a852e in PsychicEventSource::send (this=0x3ffb4628 <events>, message=<optimized out>, event=<optimized out>, id=1725798746, reconnect=0) at lib/PsychicHttp/src/PsychicEventSource.cpp:125
#4  0x401743e0 in HsHServer::sendEvent (this=<optimized out>, type=0x3f41e3a3 "time", doc=...) at src/Server/Event.cpp:71
#5  0x4017a5f1 in <lambda(int, int)>::operator() (__closure=0x3ffcb9e4, prevSec=<optimized out>, nowSec=<optimized out>) at src/Server/Server.cpp:121
#6  std::_Function_handler<void(int, int), HsHServer::handleClientTimeSync()::<lambda(int, int)> >::_M_invoke(const std::_Any_data &, int &&, int &&) (__functor=..., __args#0=<optimized out>, __args#1=<optimized out>) at c:\users\pc\.platformio\packages oolchain-xtensa-esp32@8.4.0+2021r2-patch5\xtensa-esp32-elf\include\c++\8.4.0\bits/std_function.h:297
#7  0x40197179 in std::function<void (int, int)>::operator()(int, int) const (this=0x3ffcb9e4, __args#0=<optimized out>, __args#1=<optimized out>) at c:\users\pc\.platformio\packages        oolchain-xtensa-esp32@8.4.0+2021r2-patch5\xtensa-esp32-elf\include\c++\8.4.0\bits/std_function.h:687
```

hitecSmartHome, 2024-09-09:
I have found some related crashes online.

- https://www.esp32.com/viewtopic.php?t=33571
- https://github.com/espressif/esp-idf/issues/8440
- https://github.com/espressif/esp-idf/issues/7095

They all seem to get to the same conclusion. httpd_queue is filling up and never calling the cb function. This makes it leak memory.

Quote from @martinKupec

> Hi,
I were also hit by this issue, but I cannot serialize the queue with semaphores as I need it asynchronous.
I have found the root problem. It is shortage of MEMP_NETBUF buffers in LWIP. It fails in LWIP on this line:
https://github.com/espressif/esp-lwip/blob/a45be9e438f6cf9c54ec150581819c3b95d5af6b/src/api/api_msg.c#L183
This can be solved by rising MEMP_NUM_NETBUF to some higher value (each buffer is about 36 bytes of size).
It also seems that lowering LWIP_LOOPBACK_MAX_PBUFS will at least notify the sender, that the work will never be done.
It is my understanding, that a proper fix would be to stop using socket as messaging layer and use some thread safe queue directly as all but one mallocs happen in the middle layers and are fundamentally not needed.

hitecSmartHome, 2024-09-09:
Should we create a queue blocker so it won't send more items to the queue if it too much?

hoeken, 2024-09-09:
Interesting.  Yeah, it seems like we need to add some protection logic around that call to prevent this.

BCsabaEngine, 2024-09-09:
It might be a stupid question from someone from far away, but just in case...

Did you try increase max_uri_handlers to proper value:
```
server.config.max_uri_handlers = 20; //...or more
```

hitecSmartHome, 2024-09-09:
Yeah it is something like 200 or 1000. Max open sockets is maxed too.

hoeken, 2024-09-09:
@hitecSmartHome do you have that many uri handlers?  this looks like a memory issue, and the webserver allocates memory based on that max_uri_handlers.  if you can reduce that, it might help.  (note, in 2.0 we are switching to dynamic allocation for uri handlers)

BCsabaEngine, 2024-09-09:
> @hitecSmartHome do you have that many uri handlers? this looks like a memory issue, and the webserver allocates memory based on that max_uri_handlers. if you can reduce that, it might help. (note, in 2.0 we are switching to dynamic allocation for uri handlers)

Not so much, just 20-30. We are supporting @hitecSmartHome with many files, file list is here: https://github.com/BCsabaEngine/svelteesp32/issues/30#issue-2514056419 . From this point, he reduced the file count.

hoeken, 2024-09-09:
Try temporarily reducing that for testing.  We still need to fix the underlying memory issue, but changing it might temporarily fix it until we get that in place.

I'm back from vacation now and once I get caught up with things at home, I'll focus on getting v2.0 stable and released.

hitecSmartHome, 2024-09-09:
File handlers are one thing but i have a lot of request handlers too. More than 100 in total for sure. Will check the exact number and decrease it.

hitecSmartHome, 2024-09-09:
> Not so much, just 20-30. We are supporting @hitecSmartHome with many files, file list is here: https://github.com/BCsabaEngine/svelteesp32/issues/30#issue-2514056419 . From this point, he reduced [...]

That is the v2 try. Iam using v1 for my big application but meanwhile iam testing and developing on the pioarduino fork and psychic v2. 

So I have two big projects currently. Both is the same but one is a compelte rewrite. 

The ongoing is arduino v2 and idf 4.4.7 with PsychicHttp v1. That uses a raw js frontend with webpack. No svelteesp32.

The rewrite is using pioarduino (arduino v3 and idf v5), psychic v2 and svelteeesp32. 

The first one has the event issues but that is the one currently under heavy testing.

hitecSmartHome, 2024-09-09:
Iam also using SSE and WS server in both from Psychic.

hoeken, 2024-09-09:
@hitecSmartHome for esp-idf <5.3.x we will allocate 1 endpoint per method (GET, POST, etc) + 1 per websocket endpoint.  for 5.3+ we will allocate a single endpoint for HTTP_ANY and then 1 per websocket endpoint.   Internally its then just a small std::list to store url and method.

for the end user it will be much simpler... just call server.on() and you're good.

hitecSmartHome, 2024-09-09:
Ah can't wait

mathieucarbou, 2024-09-09:
@hitecSmartHome @hoeken : did you try to set `CONFIG_HTTPD_QUEUE_WORK_BLOCKING ` ? When defined, this flag uses a semaphore to protect the code inside `httpd_queue_work `, but not only this one.

Read the manual: https://github.com/espressif/esp-idf/blob/master/components/esp_http_server/src/httpd_main.c#L19 :P

hitecSmartHome, 2024-09-09:
> @hitecSmartHome @hoeken : did you try to set `CONFIG_HTTPD_QUEUE_WORK_BLOCKING ` ? When defined, this flag uses a semaphore to protect the code inside `httpd_queue_work` (and not only this method)
> 
> Read the manual: https://github.com/espressif/esp-idf/blob/master/components/esp_http_server/src/httpd_main.c#L19 :P

Ohhhhh... that is good news. Will check it tomorrow for sure!

[12 more replies]

closed 2025-11-28 by hoeken