hoeken/PsychicHttp #98: Is there a way to invalidate cache?

hitecSmartHome, 2024-06-11:
Hi!
I wonder if is there a way to invalidate the cache controll from server side.

Currently I use `setCacheControl` set to one month but if I refresh some files or the entire flash the cache still in effect and the user sees the old web.

hitecSmartHome, 2024-06-11:
I see that the etag is coming from the file.size() but I don't think it's working

Would it work if I set the  `_last_modified` variable to a fresh string after an update?

Chris--A, 2024-06-23:
Yes, use write time, not file size as it is not an accurate test if the size doesn't change on modification, the internal file serving uses write time and I have a bunch of fixes to submit depending on how my last submissions are handled:

I use `must-revalidate` to ensure that a client will check the server for a change. Then each request I calculate the ETag from the file write time which has proven very quick, and only send the new file if the ETag is different.

I'd rather pay the price of a quick request to check opposed to the client going out of date. Once development is done I can change this, but I want guarantee I see the changes as I make them.

```C++
    server.on("/*", HTTP_GET, [](PsychicRequest *request) {
      esp_err_t result = ESP_OK;
      String path = "/www";

      path += (request->url() == "/") ? "/index.html" : request->url();

      Serial.printf("HTTP GET [%s] -> [%s] ", request->url(), path);

      std::lock_guard<std::mutex> lck(SDMutex);
      File file = SD.open(path);

      if(file){
        String etagStr(file.getLastWrite(), 10);

        if(request->header("If-None-Match").equals(etagStr)){
          Serial.println("[304]");
          result = request->reply(304);
        }else{
          PsychicFileResponse response(request, file, path);
          response.addHeader("Cache-Control", "must-revalidate");
          response.addHeader("ETag", etagStr.c_str());
          result = response.send();
        }
        file.close();
      }else{
        result = request->reply(404);
      }
      return result;
    }); 
```

hitecSmartHome, 2024-06-23:
Oh it is a really nice solution. Is this a middleware? Is this overwrites the lib default headers?

Chris--A, 2024-06-23:
> Oh it is a really nice solution. Is this a middleware? Is this overwrites the lib default headers?

No this is an endpoint. If I remember correctly, when I wrote this I did check the underlying ESP HTTP libs and the headers can be over written. You also can leave out the mutex if you do not need to synchronize access to the SD, I use it in different threads for a DB.

Endpoints are stacked, and processed one after the other, so this is the last endpoint in mine, which is why it is a generic `/*`. The more specific endpoints are run first.

hitecSmartHome, 2024-06-23:
I see. Thank you very much. Will try it out

hitecSmartHome, 2024-06-25:
I have found this function in `PsychicStaticFileHander.cpp`

```cpp
esp_err_t PsychicStaticFileHandler::handleRequest(PsychicRequest *request)
{
  if (_file == true)
  {
    //DUMP(_filename);

    //is it not modified?
    String etag = String(_file.size());
    if (_last_modified.length() && _last_modified == request->header("If-Modified-Since"))
    {
      //DUMP("Last Modified Hit");
      //TRACE();
      _file.close();
      request->reply(304); // Not modified
    }
    //does our Etag match?
    else if (_cache_control.length() && request->hasHeader("If-None-Match") && request->header("If-None-Match").equals(etag))
    {
      //DUMP("Etag Hit");
      //DUMP(etag);
      //DUMP(_cache_control);

      _file.close();

      PsychicResponse response(request);
      response.addHeader("Cache-Control", _cache_control.c_str());
      response.addHeader("ETag", etag.c_str());
      response.setCode(304);
      response.send();
    }
    //nope, send them the full file.
    else
    {
      //DUMP("No cache hit");
      //DUMP(_last_modified);
      //DUMP(_cache_control);

      PsychicFileResponse response(request, _fs, _filename);

      if (_last_modified.length())
        response.addHeader("Last-Modified", _last_modified.c_str());
      if (_cache_control.length()) {
        response.addHeader("Cache-Control", _cache_control.c_str());
        response.addHeader("ETag", etag.c_str());
      }

      return response.send();
    }
  } else {
    return request->reply(404);
  }

  return ESP_OK;
}
```

Would it be easier to just change this line

`String etag = String(_file.size());`
to that
`String etag = String(_file.getLastWrite());`

Chris--A, 2024-06-26:
> Would it be easier to just change this line
> 
> `String etag = String(_file.size());` to that `String etag = String(_file.getLastWrite());`

Sure, this is in-fact one of the changes I have lined up for a future PR. However, the issue with my code is I can't use the static file handler as I need to have control to lock the SD (mutex) as it is used by other threads. 

Also, depending on the cache control, you still won't have the client side invalidating and checking, will need to ensure that `must-revalidate` is used. I did find the client wasn't checking for changes until the cache had expired.

closed 2024-08-07 by hitecSmartHome