hoeken/PsychicHttp #100: Add in pull request guidelines and fix 'stolen'/copied code.

Chris--A, 2024-06-21:
Hi @hoeken 

I'm concerned that nothing has been done in regards to pull request #89 
I spent a lot of time researching, writing, and documenting my TemplatePrinter and the pull request was ready to merge for months.

The user who submitted #89 **copied, not merged my commits**, and then submitted my work in their own commits, which means that there is no link to the actual author of this work.

I create and submit code for open source projects to benefit others, and more importantly build a portfolio of my work that other projects and future employers can view. By allowing another user to submit my work, I have no proof that I wrote that code, and others cannot find any reference to it from my profile. 

I really must ask that you do what is right and reverse that change and submit my PR (#56) if you want the TemplatePrinter in your project. 

As my code is unrelated to #89 it should not have been included there. That PR should be reverted and they should resubmit their own work only. This will not only make your repo cleaner, but people will be able to find the PR and justification behind its implementation.

I also recommend you add in some guidelines to prevent this from happening in the future.

hoeken, 2024-06-23:
Hi @Chris--A 

I agree 100% and I'm sorry I messed this one up.  This one is definitely my fault.  I was just going through all the PRs and trying to clear the queue out.  I saw that the #89 pr included a bunch of others and my brain said that it was the easiest way.  It didn't cross my mind that you would want the original commit merged and that was an oversight.

Last week and this week are both pretty hectic for me, but as soon as I get some free time I will unwind this and merge everything in properly.  Your contributions are very much appreciated.  It wasn't an intentional slight or anything so hopefully no hard feelings. :)

hoeken, 2024-08-09:
@Chris--A I re-opened and merged your code properly + added attribution in the upcoming changelog.

closed 2024-08-09 by hoeken

mathieucarbou, 2024-08-09:
@hoeken : it is also possible to activate this option to force contributors to accept that the PR they submit is in accordance with the projet's license: 

[image]

When contributing a PR in an LGPL project, attribution is not a "right" because the work should have been submitted under the project's license.

Most projects are using SPDX generated headers so there is no attribution on any file / header but just the SPDX header.

Submitting a PR with a work requiring attribution would mean to change the license of this specific file only, and depending on the project this can be an issue, for example someone might want to submit PR with some GPL additions then preventing the library in LGPL to be used commercially.

That's why contributors should / have to accept to PR their work under a projet's license, and often, attribution is not automatic.

This is very nice from @hoeken to maintain such list.

hoeken, 2024-08-10:
yeah that's a good idea.