hoeken/PsychicHttp #101: sdcard (request info)

Red-Owl, 2024-06-24:
server.serveStatic("/", SD, "/")->setFilter(ON_STA_FILTER);    // request is "/sd" (mountpoint of sd) i use "/" root filesystem
                             ^^^
error 404

server.serveStatic("/favicon.ico", SD, "/start/favicon.ico");   this found display my icon
tnx for support

hoeken, 2024-08-09:
Can you explain this problem a little bit more?

Red-Owl, 2024-08-10:
It does not display any files present on the SD card, and returns 404

hoeken, 2024-08-10:
I just got a board with SD card today, so I can test this if we find a real bug.

```server.serveStatic("/", SD, "/")->setFilter(ON_STA_FILTER);``` 

This code will serve the file at the request url from the / directory of the sd card.  eg. request for /favicon.ico will look for sd /favicon.ico

It will also only serve the file through the STA interface (when esp32 is in client mode).

```server.serveStatic("/favicon.ico", SD, "/start/favicon.ico");```

If this code is working, then that means your web files are in sd: /start/  Try changing your first code to:

```server.serveStatic("/", SD, "/start");```

hoeken, 2024-08-11:
I just pushed 09a04c76bf9f317a54216407106f69c9072741cb to #v2-dev with example of sd card.  its working fine.  if you're still having troubles, please re-open.

closed 2024-08-11 by hoeken