hoeken/PsychicHttp #103: pioarduino: esp_https_server fails to start

mhaberler, 2024-07-20:
trying the [Arduino3 community fork ](https://github.com/pioarduino/platform-espressif32) 

both `stable` and `develop`  show [same behavior](https://github.com/mhaberler/PsychicHttp/commit/7c46223518aa41530fd7edcaf1fbe455d80c75e7)  -  esp_https_server fails to start

cert and key are definitely present

this looks like an sdkconfig issue to me, possibly around mbed_tls 

`platform = espressif32@6.7.0` works fine btw

````
jcJQBQkTfq6Z0H6OoTVi7dPs3ibQJFrtkoyvYAbyk36quBmNRjVh6rc8468bhXYr
v/t+MeGJP/0Zw8v/X2CFll96
-----END PRIVATE KEY-----
E (1871) esp_https_server: No Server certificate supplied
[  2377][E][PsychicHttpServer.cpp:84] _start(): [psychic] Server start failed (ESP_FAIL)
[  2396][E][PsychicHttpServer.cpp:151] on(): [psychic] Add endpoint failed (ESP_ERR_INVALID_ARG)
[  2405][E][PsychicHttpServer.cpp:151] on(): [psychic] Add endpoint failed (ESP_ERR_INVALID_ARG)
[  2414][E][PsychicHttpServer.cpp:151] on(): [psychic] Add endpoint failed (ESP_ERR_INVALID_ARG)
[  2422][E][PsychicHttpServer.cpp:151] on(): [psychic] Add endpoint failed (ESP_ERR_INVALID_ARG)
[  2431][E][PsychicHttpServer.cpp:151] on(): [psychic] Add endpoint failed (ESP_ERR_INVALID_ARG)
=========== After Setup Start ============
````

Jason2866, 2024-07-20:
The community Platformio build uses the officaly 3.0.3 release from espressif server.
So it is 100% same as when using ArduinoIDE. So not an Platform problem.

mhaberler, 2024-07-23:
while I have not fixed this particular issue yet, I have made progress towards HTTPS servers which work well with the Arduino3 community fork:

https://github.com/mhaberler/esp32_https_server.git
https://github.com/mhaberler/esp32_idf5_https_server_compat/tree/pioa3-test

both are based  on recent work by @jackjansen and needed a minor brushup to get past the mbed_tls API changes which happen around ESP-IDF 5.0.2

they are descendents of the abandoned https://github.com/fhessel/esp32_https_server 

however, they are NOT based on the ESP-IDF `esp_https_server` component like PsychicHTTP AFAICT

mlesniew, 2024-07-23:
Could this be the same issue as #93?

@mhaberler, can you try enabling the idf logs to see if you get any more information?  I wonder if it will also print a message saying that no certificate was supplied.

mhaberler, 2024-07-23:
it comes down to a difference in `struct httpd_ssl_config`  - V5.0.2 and later expect an additional `servercert` and `servercert_len` to be passed 

if I change [PsychicHttpsServer::listen](https://github.com/hoeken/PsychicHttp/blob/master/src/PsychicHttpsServer.cpp#L25-L36) like so:
````c
esp_err_t PsychicHttpsServer::listen(uint16_t port, const char *cert, const char *private_key) {
    this->_use_ssl = true;

    this->ssl_config.port_secure = port;
    this->ssl_config.cacert_pem = (uint8_t *)cert;
    this->ssl_config.cacert_len = strlen(cert)+1;

    this->ssl_config.servercert = (uint8_t *)cert;
    this->ssl_config.servercert_len = strlen(cert);

    this->ssl_config.prvtkey_pem = (uint8_t *)private_key;
    this->ssl_config.prvtkey_len = strlen(private_key)+1;

    return this->_start();
}
````

the https server starts up fine and accepts a connection

it fails later due to servercert expects something different than cacert_pem and parsing the cert fails, but that is to be expected

so I'm hopeful, need to read up on migration

mhaberler, 2024-07-23:
example which builds and works for Arduino2 and Arduino3: https://github.com/mhaberler/PsychicHttp-arduino3-example

PR incoming tomorrow

mhaberler, 2024-08-07:
fixed via https://github.com/hoeken/PsychicHttp/commit/98dec76f4c9dc5da1c1478ac6dddfd1449101b76 

thanks @hoeken !

closed 2024-08-07 by mhaberler