hoeken/PsychicHttp #106: is this project abandoned? please advise

mhaberler, 2024-07-26:
HI,

I note there's been very little recent activity, issues and a few PR's piling up

Like many others, I need a stable http/https/ws/wss server and I think the overall approach to use esp_http_server and esp_https_server as foundation is solid

Having used ESPAsyncWebserver  (pre-this [fork](https://github.com/mathieucarbou/ESPAsyncWebServer ) which I still have to look into) I had to eventually roll my own and pull together patches from several forks

it generally worked but there was a mushroom cloud every now and then, and a tad too often for me to consider it a solid base going forward, especially as the author was busy with other projects for the foreseeable future

I have the eerie feeling I'm heading into a similar situation with this project and I feel rather uneasy about it

Currently there is work underway to put an [MQTT broker](https://github.com/mlesniew/PicoMQTT) ontop of PsychicHTTP WSS support and I would be most concerned if the horse flakes out under us _again_.

@hoeken please advise and recommend a route for users

do you think you'll be able bring back love and time to this project?  
should it be forked and maintenance reorganized?
should we look elsewhere?

thanks in advance

Michael

hoeken, 2024-07-27:
@mhaberler I'm sorry I haven't been more active with this project, life has gotten pretty busy lately and I haven't had any cycles to devote to it.  I will try and carve out some time this week to fix that header bug, pull in the PRs and get a nice clean, stable release out.

I definitely have not abandoned it, but for the next couple months I wont have time to do any new feature related stuff on the project.  I know that's not awesome, but it's an honest look at what my schedule is looking like.  On the flip side, with the underlying ESP-IDF the library is pretty stable (when I don't mess up and accept an untested PR)

As far as forking or maintenance goes... well I'd be happy to accept some help in that regard.  Alternatively, if anyone wants to fork it, that is fine by me too.

mhaberler, 2024-07-27:
Hi Zach,

thanks for chipping in despite being busy

my concern is for continuity and a fork would have been last resort, so let's organize things while you are away

I see two options:

- bring in a few collaborators on your existing repo who can keep things afloat (messy on permissions but retains repo URL as-is)
- move PsychicHTTP to a dedicated gh organisation and bring in collaborators there (repo URL changes but possibly better long term organisation)

any others? your call

while catch-up and a release is certainly appreciated I think the setup of one or the other option above is more pressing as we can work through the PR's; occasional advice while in Starlink coverage would be great ;)

also, please enable discussions or some other vehicle like a discord channel as you see fit, so we dont have to funnel qestion through abusing issues

I'm happy to help keeping the ship afloat even I likely will not make grandiose contributions, and I understand @hitec would chip in as well

best regards

Michael

mhaberler, 2024-08-05:
people start asking me privately if my fork can be used and I have two simultaneous projects using this code

@hitecSmartHome  I suggest we take action at this point 

what about using your fork as temporary locus and inviting a few collaborators?

drop me a  line at haberlerm at gmail dot com, we can have a call lateron

hitecSmartHome, 2024-08-05:
I'm in for the maintaining. This library saving me ( and us ) a lot of trouble and time.
I did not have a fork on github, just a local copy but I'm happy to make one. But it would be way better to maintane and update the original.

mathieucarbou, 2024-08-05:
Ideally, @hoeken would need to add you as collaborators. Otherwise the same thing that happened with ESPAsyncWebsServer will happen here.

hoeken, 2024-08-05:
> ESPAsyncWebServer will happen here.

I can add you guys as collaborators.  Also, I'm finally done with boat
projects for the time being so the next week I'm planning on devoting to
this project.

----

Zach Smith

mhaberler, 2024-08-05:
please do, I'm in

I've never created a discord channel but that'd be handy to have - anybody in the know?

hitecSmartHome, 2024-08-05:
I have created one.
https://discord.com/invite/4QdP5yt6

mathieucarbou, 2024-08-05:
I've joined because I would really like to see this project become on part with the [ESPAsyncWebserver](https://github.com/mathieucarbou/ESPAsyncWebServer) fork we maintain, in terms of features, API and memory usage. 

Here are a list of things I saw so far (some might be fixed perhaps):
- No 8266 support
- No RP2040 support
- No ArduinoJson support (handlers to facilitate json support for both http api and websocket)
- No MessagePack support (handlers to facilitate json support for both http api and websocket)
- No HTTP_ANY equivalent
- No rewrite
- Not API compatible
- Async mode to be improved with more control over core, tasks, memory and such: too much memory usage by default: spawns 8 workers of 4k stack for each so 32k stack reserved and not fully used
- There has been too much API breaking and renaming occurring from minor digit to minor digit while the major digit should have been increased instead to respect the semver rules which are used for library dependency resolution

Depending on how the project goes I am willing to help (but few little time so will be as long as I will migrate some projects to support Psychic) and I would like to see if there is a way to help ESPAsyncWebserver's users migrate more easily.

hitecSmartHome, 2024-08-06:
Ws has no json support but http requests has. Iam using it like no other. As of 8266 support, I did not know about that. It is built on idf so it should not be that hard.

mhaberler, 2024-08-06:
I have more moderate goals - I need a solid basis for wss to run an MQTT broker ontop

so for me it's about navigating through the platformio and arduino unruly waters, make sure CI works and fix the ocasional bug

mathieucarbou, 2024-08-07:
I've opened https://github.com/hoeken/PsychicHttp/issues/110 to capture all the migration issues which I think are not helping users adopt Psychic.

mhaberler, 2024-08-07:
@hoeken can we lure you over to https://discord.com/invite/4QdP5yt6 ?

some guidance would be most welcome!

hitecSmartHome, 2024-08-07:
Would be great!

hoeken, 2024-08-07:
I'll pop in the chat tomorrow. I've carved out the next couple days for
this project only.

mhaberler, 2024-08-07:
great - really appreciated! 

it looks like there's quite some interested in moving this forward, plus some helping hands

delighted to see that project picking up again!

mathieucarbou, 2024-08-08:
> Honestly, 8266 is obsolete and declared EOL by espressif.

@theelims : FYI, EOL for ESP8266 is in 5 years (https://www.espressif.com/en/products/longevity-commitment)

theelims, 2024-08-08:
They extended the support from 12 to 15 years since the last time I checked. 

Anyway, since the 8266 is not supported by the underlying ESP-IDF it will require a completely different approach.