hoeken/PsychicHttp #110: ESPAsyncWebServer => PsychicHttp Migration

mathieucarbou, 2024-08-07:
I am opening this issue to list all the migration problems currently either preventing or making me abandon the move to Psychic.

mathieucarbou, 2024-08-07:
# Proposition 1: Return references instead of pointers

This specifically applies to:

```c++
    PsychicEndpoint* on(const char* uri);
    PsychicEndpoint* on(const char* uri, http_method method);
    PsychicEndpoint* on(const char* uri, PsychicHandler *handler);
    PsychicEndpoint* on(const char* uri, http_method method, PsychicHandler *handler);
    PsychicEndpoint* on(const char* uri, PsychicHttpRequestCallback onRequest);
    PsychicEndpoint* on(const char* uri, http_method method, PsychicHttpRequestCallback onRequest);
    PsychicEndpoint* on(const char* uri, PsychicJsonRequestCallback onRequest);
    PsychicEndpoint* on(const char* uri, http_method method, PsychicJsonRequestCallback onRequest);
```

That I would like to see changed into:

```c++
    PsychicEndpoint& on(const char* uri);
    PsychicEndpoint& on(const char* uri, http_method method);
    PsychicEndpoint& on(const char* uri, PsychicHandler *handler);
    PsychicEndpoint& on(const char* uri, http_method method, PsychicHandler *handler);
    PsychicEndpoint& on(const char* uri, PsychicHttpRequestCallback onRequest);
    PsychicEndpoint& on(const char* uri, http_method method, PsychicHttpRequestCallback onRequest);
    PsychicEndpoint& on(const char* uri, PsychicJsonRequestCallback onRequest);
    PsychicEndpoint& on(const char* uri, http_method method, PsychicJsonRequestCallback onRequest);
```

- This allows to chain calls with setFiter like ESPAsyncWebServer
- And why would the user be anyway in control of a pointer that is managed by the library ?

mathieucarbou, 2024-08-07:
Proposition 2: Either match ESPAsyncWS method names or provide a migration path (compile flag ?)

Examples:

send vs reply:

```c++
#ifdef PSYCHIC_TO_ESPASYNCWS
    esp_err_t send(int code) { return reply(code); }
    esp_err_t send(int code, const String& contentType = emptyString, const String& content = emptyString);
#endif
```

constructor:

```c++
PsychicHttpServer(uint16_t port = 80);
```

begin / end vs listen / stop:

```c++
#ifdef PSYCHIC_TO_ESPASYNCWS
    void begin(uint16_t port = 80) { listen(port); }
    void end() { stop(); }
#endif
```

etc...

mathieucarbou, 2024-08-07:
# Proposition 3: Provide a way to map callbacks and classes

Example:

```c++
#ifdef PSYCHIC_TO_ESPASYNCWS
  #include <espasyncws.h>
#endif
```

content:

```c++
#pragma once

#include "PsychicHttpServer.h"
#include "PsychicWebHandler.h"
#include "PsychicRequest.h"

using AsyncWebServer = PsychicHttpServer;
using AsyncCallbackWebHandler = PsychicWebHandler;
using AsyncWebServerRequest = PsychicRequest;
```

Note: I am not sure this is the right way to do it, it could also be done within the library classes, example right now I am trying to migrate by using that in a library:

```c++
#ifdef ESPCONNECT_USE_PSYCHIC
    using Server = PsychicHttpServer;
    using WebHandler = PsychicWebHandler;
#else
    using Server = AsyncWebServer;
    using WebHandler = AsyncCallbackWebHandler;
#endif
```

So this is scoped to only my classes, but if this concept works, why not extending it globally and have Psychic provide a mapping class.

mathieucarbou, 2024-08-07:
# Proposition 4: make handlers return `void:

```c++
typedef std::function<void(PsychicRequest *request)> PsychicHttpRequestCallback;
typedef std::function<void(PsychicRequest *request, JsonVariant &json)> PsychicJsonRequestCallback;
```

I don't understand why a handler would return an internal ESP status code: a handler is linked to serving a request and any status code within this context should lead to an http status code answer.

Those are acceptable:

```c++
    esp_err_t redirect(const char *url);
    esp_err_t reply(int code);
    esp_err_t reply(const char *content);
    esp_err_t reply(int code, const char *contentType, const char *content);
```

Because they are linked to the underlying API:

```
*  - ESP_OK : On successfully sending the response packet
 *  - ESP_ERR_INVALID_ARG : Null request pointer
 *  - ESP_ERR_HTTPD_RESP_HDR    : Essential headers are too large for internal buffer
 *  - ESP_ERR_HTTPD_RESP_SEND   : Error in raw send
 *  - ESP_ERR_HTTPD_INVALID_REQ : Invalid request
 */
esp_err_t httpd_resp_send(httpd_req_t *r, const char *buf, ssize_t buf_len);
```

And one would want to know if his request is sent correctly, and **act upon this result within the handler.**

But what is the use case for having a handler returning an error code ?

hoeken, 2024-08-11:
created individual threads for these for easier discussion

closed 2024-08-11 by hoeken