hoeken/PsychicHttp #118: Support multipart/form-data

mathieucarbou, 2024-08-07:
A lof of ajax request are using multipart/form-data to send forms dynamically.

ESPAsyncWebSerevr supports that well with `request->getParam(key, true)` but Psychic only supports `application/x-www-form-urlencoded`

```
Content-Type: multipart/form-data; boundary=----WebKitFormBoundaryAxovNqdMd5Jtt9aB
```

```
------WebKitFormBoundaryAxovNqdMd5Jtt9aB
Content-Disposition: form-data; name="ssid"

------WebKitFormBoundaryAxovNqdMd5Jtt9aB
Content-Disposition: form-data; name="password"

------WebKitFormBoundaryAxovNqdMd5Jtt9aB
Content-Disposition: form-data; name="ap_mode"

true
------WebKitFormBoundaryAxovNqdMd5Jtt9aB--
```

hoeken, 2024-08-09:
Can you either get me a curl oneliner or modify the platformio example to demonstrate the desired functionality?

hoeken, 2024-08-10:
Okay, this command will send a multipart form and dump the full session.

```curl -F "param1=multi" -F "param2=part" http://psychic.local/post --trace-ascii -```

mathieucarbou, 2024-08-10:
> Okay, this command will send a multipart form and dump the full session.
> 
> `curl -F "param1=multi" -F "param2=part" http://psychic.local/post --trace-ascii -`

```bash
❯  curl -F "param1=multi" -F "param2=part" http://webhook.site/a7bce3a7-899b-4e28-9f1b-ec7da358d251 --trace-ascii -
== Info: Host webhook.site:80 was resolved.
== Info: IPv6: (none)
== Info: IPv4: 178.63.67.153, 178.63.67.106, 46.4.105.116
== Info:   Trying 178.63.67.153:80...
== Info: Connected to webhook.site (178.63.67.153) port 80
=> Send header, 225 bytes (0xe1)
0000: POST /a7bce3a7-899b-4e28-9f1b-ec7da358d251 HTTP/1.1
0035: Host: webhook.site
0049: User-Agent: curl/8.9.1
0061: Accept: */*
006e: Content-Length: 263
0083: Content-Type: multipart/form-data; boundary=--------------------
00c3: ----CLhjWCNNKe3FBYwEUcXMjj
00df: 
=> Send data, 263 bytes (0x107)
0000: --------------------------CLhjWCNNKe3FBYwEUcXMjj
0032: Content-Disposition: form-data; name="param1"
0061: 
0063: multi
006a: --------------------------CLhjWCNNKe3FBYwEUcXMjj
009c: Content-Disposition: form-data; name="param2"
00cb: 
00cd: part
00d3: --------------------------CLhjWCNNKe3FBYwEUcXMjj--
== Info: upload completely sent off: 263 bytes
<= Recv header, 17 bytes (0x11)
0000: HTTP/1.1 200 OK
<= Recv header, 15 bytes (0xf)
0000: server: nginx
<= Recv header, 40 bytes (0x28)
0000: content-type: text/html; charset=UTF-8
<= Recv header, 28 bytes (0x1c)
0000: transfer-encoding: chunked
<= Recv header, 52 bytes (0x34)
0000: x-request-id: 7d87782d-09c0-4c55-a76e-b8fa8ded62c8
<= Recv header, 50 bytes (0x32)
0000: x-token-id: a7bce3a7-899b-4e28-9f1b-ec7da358d251
<= Recv header, 34 bytes (0x22)
0000: cache-control: no-cache, private
<= Recv header, 37 bytes (0x25)
0000: date: Sat, 10 Aug 2024 19:53:15 GMT
<= Recv header, 2 bytes (0x2)
0000: 
<= Recv data, 156 bytes (0x9c)
0000: 91
0004: This URL has no default content configured. <a href="https://web
0044: hook.site/#!/view/a7bce3a7-899b-4e28-9f1b-ec7da358d251">View in 
0084: Webhook.site</a>.
0097: 0
009a: 
This URL has no default content configured. <a href="https://webhook.site/#!/view/a7bce3a7-899b-4e28-9f1b-ec7da358d251">View in Webhook.site</a>.== Info: Connection #0 to host webhook.site left intact
```

I am using curl with -F to test with my app also.

hoeken, 2024-08-11:
This is now fixed :)

closed 2024-08-11 by hoeken