hoeken/PsychicHttp #150: v2 - Return references instead of pointers

hoeken, 2024-08-11:
# Proposition 1: Return references instead of pointers

This specifically applies to:

```c++
    PsychicEndpoint* on(const char* uri);
    PsychicEndpoint* on(const char* uri, http_method method);
    PsychicEndpoint* on(const char* uri, PsychicHandler *handler);
    PsychicEndpoint* on(const char* uri, http_method method, PsychicHandler *handler);
    PsychicEndpoint* on(const char* uri, PsychicHttpRequestCallback onRequest);
    PsychicEndpoint* on(const char* uri, http_method method, PsychicHttpRequestCallback onRequest);
    PsychicEndpoint* on(const char* uri, PsychicJsonRequestCallback onRequest);
    PsychicEndpoint* on(const char* uri, http_method method, PsychicJsonRequestCallback onRequest);
```

That I would like to see changed into:

```c++
    PsychicEndpoint& on(const char* uri);
    PsychicEndpoint& on(const char* uri, http_method method);
    PsychicEndpoint& on(const char* uri, PsychicHandler *handler);
    PsychicEndpoint& on(const char* uri, http_method method, PsychicHandler *handler);
    PsychicEndpoint& on(const char* uri, PsychicHttpRequestCallback onRequest);
    PsychicEndpoint& on(const char* uri, http_method method, PsychicHttpRequestCallback onRequest);
    PsychicEndpoint& on(const char* uri, PsychicJsonRequestCallback onRequest);
    PsychicEndpoint& on(const char* uri, http_method method, PsychicJsonRequestCallback onRequest);
```

- This allows to chain calls with setFiter like ESPAsyncWebServer
- And why would the user be anyway in control of a pointer that is managed by the library ?

mathieucarbou, 2024-08-11:
@hoeken : let's close this one. I found a cleaner solution by adding decorator classes, which does not touch the current API: https://github.com/hoeken/PsychicHttp/pull/115/files

closed 2024-08-11 by mathieucarbou

hoeken, 2024-08-11:
thats awesome!