hoeken/PsychicHttp #152: v2 - Provide a way to map callbacks and classes

hoeken, 2024-08-11:
Example:

```c++
#ifdef PSYCHIC_TO_ESPASYNCWS
  #include <espasyncws.h>
#endif
```

content:

```c++
#pragma once

#include "PsychicHttpServer.h"
#include "PsychicWebHandler.h"
#include "PsychicRequest.h"

using AsyncWebServer = PsychicHttpServer;
using AsyncCallbackWebHandler = PsychicWebHandler;
using AsyncWebServerRequest = PsychicRequest;
```

Note: I am not sure this is the right way to do it, it could also be done within the library classes, example right now I am trying to migrate by using that in a library:

```c++
#ifdef ESPCONNECT_USE_PSYCHIC
    using Server = PsychicHttpServer;
    using WebHandler = PsychicWebHandler;
#else
    using Server = AsyncWebServer;
    using WebHandler = AsyncCallbackWebHandler;
#endif
```

So this is scoped to only my classes, but if this concept works, why not extending it globally and have Psychic provide a mapping class.

hoeken, 2024-08-11:
I like approach #1 - keep everything centrally located and keep the Psychic classes tidy without a lot of #ifdefs scattered around.

mathieucarbou, 2024-08-11:
> I like approach #1 - keep everything centrally located and keep the Psychic classes tidy without a lot of #ifdefs scattered around.

that's my goal, but for some exceptions it won't be possible... I.e. not worth redefining a whole decorator class if only 1 or 2 methods names have changd

hoeken, 2024-08-11:
@mathieucarbou if its just method names, or method parameters and not return values, we can add the compatibility functions directly to the class, no problem.