hoeken/PsychicHttp #153: v2 - make handlers return void

hoeken, 2024-08-11:
# Proposition 4: make handlers return void:

```c++
typedef std::function<void(PsychicRequest *request)> PsychicHttpRequestCallback;
typedef std::function<void(PsychicRequest *request, JsonVariant &json)> PsychicJsonRequestCallback;
```

I don't understand why a handler would return an internal ESP status code: a handler is linked to serving a request and any status code within this context should lead to an http status code answer.

Those are acceptable:

```c++
    esp_err_t redirect(const char *url);
    esp_err_t reply(int code);
    esp_err_t reply(const char *content);
    esp_err_t reply(int code, const char *contentType, const char *content);
```

Because they are linked to the underlying API:

```
*  - ESP_OK : On successfully sending the response packet
 *  - ESP_ERR_INVALID_ARG : Null request pointer
 *  - ESP_ERR_HTTPD_RESP_HDR    : Essential headers are too large for internal buffer
 *  - ESP_ERR_HTTPD_RESP_SEND   : Error in raw send
 *  - ESP_ERR_HTTPD_INVALID_REQ : Invalid request
 */
esp_err_t httpd_resp_send(httpd_req_t *r, const char *buf, ssize_t buf_len);
```

And one would want to know if his request is sent correctly, and **act upon this result within the handler.**

But what is the use case for having a handler returning an error code ?

hoeken, 2024-08-11:
I'm looking through the esp-idf code here, and what is boils down to is returning anything other than ESP_OK from the handler will close the connection.

So, the question is:  is the ability to close the connection from the requestCallback more important than espasyncws compatability?  Is anyone even using it this way?  Is this a good idea?

Here's a potential good solution:

* We could store the error status in the request object.  defaults to ESP_OK, returns that in the top level handler.
* It would allow us to return void and maintain compatibility
* It would be a fairly simple change if anyone is relying on that error code
* We should probably look into existing uses of this return code - should they error and close the connection, or would it be better to send an http error code and return ESP_OK

mathieucarbou, 2024-08-11:
> I'm looking through the esp-idf code here, and what is boils down to is returning anything other than ESP_OK from the handler will close the connection.

I didn't know about that. Why would a handler close a connection ? A connection is linked to a client, not a request...

Connection closing can also be achieved with `response->addHeader("Connection", "close");`, which is what ESPAsyncWS is adding as header on the responses...

I agree with you, let's leave the handlers as-is.

We can even close this one because I will go with a decorator approach so API deos not need to be strictly compatible with ESPAsyncWS

closed 2024-08-11 by mathieucarbou