hoeken/PsychicHttp #156: defer registration of endpoint

hoeken, 2024-08-11:
defer registration of endpoint: we must be able to call on() to add endpoint before or after begin() is called, and more importantly,  we must be able to call begin() followed by end() followed by begin() again and things get back to where they where with the same endpoints.

proddy, 2024-08-11:
and make sure calling on() before the network is started (wifi/ETH) doesn't cause the crash as it did in v1

hoeken, 2024-08-11:
@mathieucarbou tested starting/stopping server every 5 seconds, seemed to work.

@proddy I added server.isConnected() and a check to server.start() which should prevent crashes.  Honestly I dont know if this approach is the best way or not, but it seems to work from my testing.  There's also ethernet support, but I don't have any ethernet boards so its a shot in the dark.

closed 2024-08-11 by hoeken

woodenplastic, 2024-08-11:
Hi, i have olimex Poe Iso boards here. If it helps to run tests for you...

hoeken, 2024-08-12:
@woodenplastic if you could test against the v2-dev branch that would be great. you will need to define PSY_ENABLE_ETHERNET to turn on that automatic check on server start.  Make sure to not initialize the wifi or softap.  It should startup with ethernet, and error out with no local ip address.