hoeken/PsychicHttp #162: Change the license

dzungpv, 2024-08-14:
I wonder why you set the license LGPL? This license prevent use the library in commercial project without release the project source code. LGPL allow to link the lib without using the same license but with static file, it is nearly impossible with ESP32 project.
The original ESP http server using Apache-2.0 license, ESPAsyncServer not using any license as you mention in the commit https://github.com/hoeken/PsychicHttp/commit/e3225537252205f2f1075ccfdfdb391f72f25861
So better change license to MIT or Apache license.

hoeken, 2024-08-14:
well, its mainly because the original library i forked (ArduinoMongoose) was LGPL.  i'll email the guy and see if he's okay with me changing it.  pretty sure there won't be any issues.

don't worry, you can definitely use this library in a closed source, commercial project.  nobody will be coming after you :)

dzungpv, 2024-08-14:
> well, its mainly because the original library i forked (ArduinoMongoose) was LGPL. i'll email the guy and see if he's okay with me changing it. pretty sure there won't be any issues.
>
> don't worry, [...]

Yes, but have a commercial ready better.

commit 67bc0b10 (2024-08-14): Changed license to MIT. fixes #162
closed 2024-08-14 by hoeken