hoeken/PsychicHttp #163: v1.2.0 - Linking Issue

DMSimone, 2024-08-14:
Upgrading from v1.1.0 to v.1.2.0 I get multiple linking errors. i.e.:
../lib/gcc/xtensa-esp32s3-elf/8.4.0/../../../../xtensa-esp32s3-elf/bin/ld.exe: .pio\build\seeed_xiao_esp32s3\src\main.cpp.o:(.literal.exit._GLOBAL__sub_D_local_hostname+0x2c): undefined reference to PsychicHttpsServer::~PsychicHttpsServer()'

This error gets repeated for all PsychicHttp related uses. Not sure where to start with diagnosing this. The project is fairly simple, just an HTTPS server running WSS. All that aside this has been an amazing library compared to all of the other options I've tested. Keep up the great work.

The platformio.ini is:
[env:seeed_xiao_esp32s3]
platform = espressif32
board = seeed_xiao_esp32s3
framework = arduino
board_build.filesystem = littlefs

monitor_speed = 115200
monitor_filters = esp32_exception_decoder, direct

build_flags =
	-D ARDUINOTRACE_ENABLE=0
	-D ASYNC_WORKER_COUNT=20

lib_deps = 
	bblanchon/ArduinoJson@7.1.0
	hoeken/PsychicHttp@1.1.0

hoeken, 2024-08-14:
could this be the issue?  v1.1.0 in the config?

```
lib_deps =
bblanchon/ArduinoJson@7.1.0
hoeken/PsychicHttp@1.1.0
```

also, make sure you are including both the library and the https server:

```
#include <PsychicHttp.h>
#include <PsychicHttpsServer.h>
```

mhaberler, 2024-08-14:
@DMSimone 
- post a repo with a reproducible example
- nail the platform to a specific version - `platform = espressif32` will get you whatever happens to be on your machine

example:
````
;platform = espressif32 @ 6.8.1
platform = https://github.com/pioarduino/platform-espressif32/releases/download/51.03.04/platform-espressif32.zip
````

DMSimone, 2024-08-14:
> could this be the issue?  v1.1.0 in the config?
> 
> ```
> lib_deps =
> bblanchon/ArduinoJson@7.1.0
> hoeken/PsychicHttp@1.1.0
> ```
> 
> also, make sure you are including both the library and the https server:
> 
> ```
> #include <PsychicHttp.h>
> #include <PsychicHttpsServer.h>
> ```

Compiles fine with 1.1.0 it faults on 1.2.0.
I'll strip down the program and share the repo when I get a chance.

DMSimone, 2024-08-14:
I don't think this is worth a repo share over. In short I can repeat the issue with the following main.cpp file:
#include <PsychicHttp.h>
#include <PsychicHttpsServer.h>

PsychicHttpsServer server;

void setup() {}
void loop() {}

The linking error occurs compiling the above at version 1.2.0, not as 1.1.0. If swapping PsychicHttpsServer for PsychicHttpServer no error occurs no matter the version.

For additional reference the platformio.ini is:

[env:seeed_xiao_esp32s3]
platform = espressif32
board = seeed_xiao_esp32s3
framework = arduino

lib_deps = 
	hoeken/PsychicHttp@1.2.0

commit 9b5966c9 (2024-08-14): added <sdkconfig.h> which adds missing #define.  fixes #163
closed 2024-08-14 by hoeken