hoeken/PsychicHttp #171: Allow reply with JsonDocument or JsonObject or JsonArray?

hitecSmartHome, 2024-08-17:
I wonder if it would be doable to reply directly with a JsonDocument in a cb.
Here is one of my endpoint controller method.

```cpp
esp_err_t DBController::getDirTree(PsychicRequest *req){
    if( !req->hasParam("path") ){
        return req->reply(400, "text/html", "Missing path");
    }

    JsonDocument tree;
    std::string jsonString;

    String path = req->getParam("path")->value();
    bool result = db.getDirTree(path.c_str(), tree);
    size_t serializedSize = serializeJson(tree, jsonString);

    if( !result || serializedSize <= 0 ){
        return req->reply(404, "text/html", "Failed to get directory tree");
    }

    return req->reply(200, "application/json", jsonString.c_str());
}
```

### Would be good to do it like this

```cpp
esp_err_t DBController::getDirTree(PsychicRequest *req){
    if( !req->hasParam("path") ){
        return req->reply(400, "text/html", "Missing path");
    }
    JsonDocument tree;
    String path = req->getParam("path")->value();
    if( !db.getDirTree(path.c_str(), tree) ){
        return req->reply(404, "text/html", "Failed to get directory tree");
    }

    return req->reply(200, "application/json", tree); // reply with JsonDocument directly
}
```

mathieucarbou, 2024-08-17:
That is definitely doable.  was think about the same. With the new refactoring in V2, it would help avoid dealing with a decorator class. The content type could be made optional, and if provided, could help switch between message pack and normal json.

hitecSmartHome, 2024-08-17:
I'm thinking about something like this

```cpp
esp_err_t PsychicRequest::reply(int code, JsonDocument &doc)
{
  PsychicResponse response(this);

  response.setCode(code);
  response.setContentType("application/json");
  response.setContent(doc);

  return response.send();
}
```

```cpp
void PsychicResponse::setContent(JsonDocument &doc)
{
  serializeJson(doc, _body);
  setContentLength(strlen(_body));
}
```

hitecSmartHome, 2024-08-17:
We need to call only this `return req->reply(200, tree);`. Since it's a JsonDocument the content type is self-explanatory

mathieucarbou, 2024-08-17:
> We need to call only this `return req->reply(200, tree);`. Since it's a JsonDocument the content type is self-explanatory

Ct has to be optional since the same JsonVariant object can be serialised as message pack also ;-)

hitecSmartHome, 2024-08-17:
Yeah, I see