hoeken/PsychicHttp #173: Authentication Failure Content Incorrect

rdnn, 2024-08-20:
https://github.com/hoeken/PsychicHttp/blob/b719b484d79860fd92623eecd9ed337b63d51905/src/PsychicRequest.cpp#L507

I believe this should be:

```C++
response.setContent(authFailMsg);
```

rdnn, 2024-08-20:
> The code is right, the bug is elsewhere.

`authFailMsg` is never used in `PsychicRequest::requestAuthentication()`. The code, as is, uses `authStr` for the header and content, which I don't believe is the desired behavior.

mathieucarbou, 2024-08-20:
> > The code is right, the bug is elsewhere.
> 
> `authFailMsg` is never used in `PsychicRequest::requestAuthentication()`. The code, as is, uses `authStr` for the header and content, which I don't believe is the desired behavior.

Oh sorry! Yes you are right for the usage. The description of the failure should be returned.
Though it requires the pointer to still be valid, plus, content type is fixed to htm.
This part of the code will be changed to a middleware in v2 which is more flexible and which can be applied to several endpoints.

hoeken, 2025-11-28:
Coming back to this after a long break and cleaning up issues.  Please re-open if its still a problem.

closed 2025-11-28 by hoeken