hoeken/PsychicHttp #176: Crash with simple code

gergocs, 2024-09-02:
With this simple code:
```
    PsychicHttpServer server = PsychicHttpServer();
    Serial.begin(9600);
    WiFiClass::mode(WIFI_AP_STA);
    WiFi.begin(SSID, PASSWORD);
    while (WiFi.status() != WL_CONNECTED) {
        delay(500);
    }

    delay(1000);

    server.listen(80);
```
the ESP crashes.
```
Core  0 register dump:
MEPC    : 0x42006a46  RA      : 0x42006a40  SP      : 0x40814f00  GP      : 0x4080a2e0  
TP      : 0x4080e144  T0      : 0x408058dc  T1      : 0x40805c92  T2      : 0xffffffff  
S0/FP   : 0x00000000  S1      : 0x00000000  A0      : 0x00000001  A1      : 0x00000001  
A2      : 0x00000004  A3      : 0x00000000  A4      : 0x40811000  A5      : 0x20001000  
A6      : 0x00000005  A7      : 0x00000009  S2      : 0x00000000  S3      : 0x00000000  
S4      : 0x00000000  S5      : 0x00000000  S6      : 0x00000000  S7      : 0x00000000  
S8      : 0x00000000  S9      : 0x00000000  S10     : 0x00000000  S11     : 0x00000000  
T3      : 0xffffffff  T4      : 0xffffffff  T5      : 0xffffffff  T6      : 0xffffffff  
MSTATUS : 0x00001881  MTVEC   : 0x40800001  MCAUSE  : 0x00000005  MTVAL   : 0x00000004  
MHARTID : 0x00000000  

Stack memory�Guru Meditation Error: Core  0 panic'ed (Load access fault). Exception was unhandled.
```
(ESP32-C6-devkitM-1)
If i delete the `server.listen(80)` line then it works. Also it crashes instantly doesn't wait for `delay(1000)`

hitecSmartHome, 2024-09-02:
Is this an esp8266?

mathieucarbou, 2024-09-02:
@gergocs : can you please decode the stack trace with your elf file ?
=> https://maximeborges.github.io/esp-stacktrace-decoder/

> Also it crashes instantly doesn't wait for delay(1000)

That is weird...

Please upload your ino file and tell us which version you are using. I have a C6 at home so I should be able to check.

gergocs, 2024-09-03:
I'm using the main branch so the latest.

The full output of crash:
```
Guru Meditation Error: Core  0 panic'ed (Load access fault). Exception was unhandled.

Core  0 register dump:
MEPC    : 0x42006396  RA      : 0x42006390  SP      : 0x40814a60  GP      : 0x40809e40  
TP      : 0x4080dd68  T0      : 0x40805438  T1      : 0x40805d50  T2      : 0xffffffff  
S0/FP   : 0x00000000  S1      : 0x00000000  A0      : 0x00000001  A1      : 0x00000001  
A2      : 0x00000004  A3      : 0x00000000  A4      : 0x40811000  A5      : 0x20001000  
A6      : 0x00000005  A7      : 0x00000009  S2      : 0x00000000  S3      : 0x00000000  
S4      : 0x00000000  S5      : 0x00000000  S6      : 0x00000000  S7      : 0x00000000  
S8      : 0x00000000  S9      : 0x00000000  S10     : 0x00000000  S11     : 0x00000000  
T3      : 0xffffffff  T4      : 0xffffffff  T5      : 0xffffffff  T6      : 0xffffffff  
MSTATUS : 0x00001881  MTVEC   : 0x40800001  MCAUSE  : 0x00000005  MTVAL   : 0x00000004  
MHARTID : 0x00000000  

Stack memory:
40814a60: 0x00000000 0x00000000 0x00000000 0x4207f9b0 0x00000000 0x00001388 0x00000000 0x00000001
40814a80: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
40814aa0: 0x00000000 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xbaad5678 0x00000160
40814ac0: 0xabba1234 0x00000154 0x40814990 0x724079c8 0x4080c76c 0x4087d620 0x40814ac8 0x4080c764
40814ae0: 0x00000018 0xe2f7130b 0x5d7fcaaf 0x40814ac8 0x00000000 0x00000001 0x408138b8 0x6e69616d
40814b00: 0xabfd6900 0xc9324314 0x00d7d45e 0x00000000 0x40814ab0 0x00000001 0x00000000 0x00000000
40814b20: 0x00000000 0x00000000 0x40811d3c 0x40811da4 0x40811e0c 0x00000000 0x00000000 0x00000001
40814b40: 0x00000000 0x00000000 0x00000000 0x42076796 0x00000000 0x00000000 0x00000000 0x00000000
40814b60: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
40814b80: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
40814ba0: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
40814bc0: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
40814be0: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
40814c00: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x02000000 0xbaad5678
40814c20: 0x0000090c 0xabba1234 0x00000900 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5
40814c40: 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5
40814c60: 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5
40814c80: 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5
40814ca0: 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5
40814cc0: 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5
40814ce0: 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5
40814d00: 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5
40814d20: 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5
40814d40: 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5
40814d60: 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5
40814d80: 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5
40814da0: 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5
40814dc0: 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5
40814de0: 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5
40814e00: 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5
40814e20: 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5
40814e40: 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5

ELF file SHA256: eb93f17922b374d2
```
With `Serial.being(9600)` the Stack memory doesn't show up.
```
0x40805438: vPortEnterCritical at /COMPONENT_FREERTOS_DIR/FreeRTOS-Kernel/portable/riscv/port.c:323
0x40805d50: mmu_ll_get_page_size at /COMPONENT_HAL_DIR/esp32c6/include/hal/mmu_ll.h:67
0x40805d50: mmu_hal_pages_to_bytes at /COMPONENT_HAL_DIR/mmu_hal.c:36
```
I'm using platformio with arduino so I have main.cpp
main.cpp:
```
#include <PsychicHttpServer.h>
#include <PsychicEventSource.h>
#include <WiFi.h>

PsychicHttpServer server = PsychicHttpServer();
PsychicEventSource eventSource = PsychicEventSource();

void setup() {
    WiFiClass::mode(WIFI_STA);
    WiFi.begin(SSID, PASSWORD);
    while (WiFi.status() != WL_CONNECTED) {
        delay(500);
    }

    delay(1000);

    server.config.max_uri_handlers = 20;
    server.listen(80);
}

void loop() {
}

```
(The SSID and password are created with `#define ""`

hitecSmartHome, 2024-09-03:
I'm sure you left out something important

gergocs, 2024-09-03:
platformio.ini:
```
[env]
platform = espressif32
framework = arduino
build_src_filter =
    -<*>
[env:main]
build_src_filter = +<tmp>
board = esp32-c6-devkitc-1
build_flags = -fexceptions
build_unflags = -fno-exceptions
lib_ldf_mode = deep
lib_deps =
    https://github.com/hoeken/PsychicHttp.git
```
tmp folder only contains a main.cpp.

gergocs, 2024-09-03:
Update:
After removing this line from `esp_err_t PsychicHttpServer::_start()` methode
```
  // Register handler
  ret = httpd_register_err_handler(server, HTTPD_404_NOT_FOUND, PsychicHttpServer::notFoundHandler);
  if (ret != ESP_OK)
    ESP_LOGE(PH_TAG, "Add 404 handler failed (%s)", esp_err_to_name(ret));
```
it doesn't crash, but now if I try to add a new handler for the server it crashes.
It looks like the issue is with `httpd_register_uri_handler` and `httpd_register_err_handler`
Also one thing to add is the `server` property is not null so it is initialized.

gergocs, 2024-09-03:
Update:
I have deleted platformio and reinstalled it. And some kind of reason it works. Don't ask me why.

hitecSmartHome, 2024-09-03:
Probably some cache issue. First thing I do in these mysterious cases is to do a full clean and restart ws code. If it does not help delete any cache folder and pio folder

closed 2024-11-08 by gergocs