hoeken/PsychicHttp #184: Stuck loading async files

johnnytolengo, 2024-09-22:
Hi, 
I have running a web server (esp32, Psychihttp verison 1.2.1, Using latest Chrome) on static files (js chunk), and sometimes it stuck when loading 
when it happens those request are forever in "Pending". 
After I refresh the browser all loads ok, because  the already loaded are cached in the browser and the missing are loaded again.
Any idea how to fix that?

[image]

hitecSmartHome, 2024-09-22:
Increase max open tcp socket in the server config struct

johnnytolengo, 2024-09-22:
config.max_open_sockets=10;
Fixed that issue.
Thank you!

johnnytolengo, 2024-09-22:
[image]

hitecSmartHome, 2024-09-22:
Np. You have to read a little bit in the IDF docs and on the internet so you can understand more about it. You will face other challenges down the road as your frontend develops.

hitecSmartHome, 2024-09-23:
@johnnytolengo  `lru_purge_enable` and `enable_so_linger` options are also useful.
Please close this issue if it is resolved.

closed 2024-09-23 by johnnytolengo