hoeken/PsychicHttp #193: WARNING: "MAX" redefined error

CelliesProjects, 2024-10-11:
Great library! I am a hardcore AsyncWebServer user and getting my feet wet with this lib. 
I like it. 

Any chance you wrap up the ESP-IDF websocket client like this lib?

Not  a big issue but I get the following error: 

```bash
.pio/libdeps/lolin_c3_mini/PsychicHttp/src/async_worker.cpp:127: warning: "MAX" redefined
 #define MAX(a, b) (((a) > (b)) ? (a) : (b))
 
In file included from /home/cellie/.platformio/packages/framework-arduinoespressif32/tools/sdk/esp32c3/include/riscv/include/riscv/csr.h:37,
                 from /home/cellie/.platformio/packages/framework-arduinoespressif32/tools/sdk/esp32c3/include/hal/esp32c3/include/hal/cpu_ll.h:16,
                 from /home/cellie/.platformio/packages/framework-arduinoespressif32/tools/sdk/esp32c3/include/hal/include/hal/cpu_hal.h:16,
                 from /home/cellie/.platformio/packages/framework-arduinoespressif32/tools/sdk/esp32c3/include/esp_hw_support/include/esp_cpu.h:14,
                 from /home/cellie/.platformio/packages/framework-arduinoespressif32/tools/sdk/esp32c3/include/esp_hw_support/include/soc/cpu.h:14,
                 from /home/cellie/.platformio/packages/framework-arduinoespressif32/tools/sdk/esp32c3/include/esp_hw_support/include/soc/spinlock.h:11,
                 from /home/cellie/.platformio/packages/framework-arduinoespressif32/tools/sdk/esp32c3/include/freertos/port/riscv/include/freertos/portmacro.h:38,
                 from /home/cellie/.platformio/packages/framework-arduinoespressif32/tools/sdk/esp32c3/include/freertos/include/freertos/portable.h:51,
                 from /home/cellie/.platformio/packages/framework-arduinoespressif32/tools/sdk/esp32c3/include/freertos/include/freertos/FreeRTOS.h:63,
                 from /home/cellie/.platformio/packages/framework-arduinoespressif32/cores/esp32/Arduino.h:33,
                 from .pio/libdeps/lolin_c3_mini/PsychicHttp/src/PsychicCore.h:32,
                 from .pio/libdeps/lolin_c3_mini/PsychicHttp/src/async_worker.h:4,
                 from .pio/libdeps/lolin_c3_mini/PsychicHttp/src/async_worker.cpp:1:
/home/cellie/.platformio/packages/toolchain-riscv32-esp@8.4.0+2021r2-patch5/riscv32-esp-elf/sys-include/sys/param.h:28: note: this is the location of the previous definition
 #define MAX(a,b) ((a) > (b) ? (a) : (b))
```

CelliesProjects, 2024-10-11:
BTW only on the esp32-c3, the esp32-s2 seems not to have the issue.

hitecSmartHome, 2024-10-11:
Are you using the v1 or the v2 branch?

CelliesProjects, 2024-10-12:
Using the master branch.

```bash
lib_deps =
    adafruit/Adafruit SHT31 Library@^2.2.2
    https://github.com/jcomas/S8_UART#4be0ce6
    hoeken/PsychicHttp
```

hoeken, 2025-11-28:
Coming back to this after a long break and cleaning up issues.  Please re-open if its still a problem.

closed 2025-11-28 by hoeken