hoeken/PsychicHttp #195: Use .serveStatic and  .on for same URI

MicDG, 2024-10-23:
Hello, doing something like this:

```
httpServer.on("/index.html", HTTP_POST, [](PsychicRequest *request) {
    // Do sth
});
httpServer.serveStatic("/index.html", LittleFS, "/index.html");
```

causes the server to return an error like "httpd_uri: Method '1' not allowed for URI". However doing this works:

```
httpServer.on("/index.html", HTTP_POST, [](PsychicRequest *request) {
    // Do sth
});
httpServer.on("/index.html", HTTP_GET, [](PsychicRequest *request) {
    // Return file
});
```

I guess this is due to the fact that .serveStatic is an handler and doesn't register any endpoint. Is there a way around this, beyond using two endpoints (it's just for brevity I guess, not having to reimplement a function to serve the file).

MicDG, 2024-10-23:
Sorry, just saw this can be used:
```

String filename = "/path/to/file";
PsychicFileResponse response(request, LittleFS, filename);

return response.send();
```

However, maybe this could be added in the ReadMe notes just for clarity.

hitecSmartHome, 2024-10-30:
After IDF 5.3 there will be `HTTP_ANY` support

hoeken, 2024-10-30:
Actually in psychic 2.0 we add HTTP_ANY support, and with 5.3 it's much
cleaner internally.  :)

mathieucarbou, 2024-10-30:
@hitecSmartHome yes!

hitecSmartHome, 2024-10-30:
Wow. I must try it