hoeken/PsychicHttp #196: Compatibility with Arduino Uno R4 WiFi

jeremiahrose, 2024-10-30:
Looking at the [listing](https://docs.arduino.cc/libraries/psychichttp/) for this library, it appears that the Arduino Uno R4 WiFi is not supported?

Are there plans to support this board at all?

jeremiahrose, 2024-10-30:
Using the following test sketch in the Arduino IDE:

```
#include <Arduino.h>
#include <WiFiS3.h>
#include <PsychicHttp.h>

#define WIFI_SSID "TEST_SSID"
#define WIFI_PASSWORD "TEST_PASSWORD"

PsychicHttpServer server;

void setup() {
  Serial.begin(115200);
  connectToWiFi();
  startServer();
}

void loop() {
}

void connectToWiFi() {
  Serial.println("connectToWiFi");
  WiFi.begin(WIFI_SSID, WIFI_PASSWORD);
  waitForConnection();
}

void waitForConnection() {
  Serial.println("waitForConnection");
  while (WiFi.status() != WL_CONNECTED) {
    delay(500);
  }
  displayConnectionDetails();
}

void displayConnectionDetails() {
  Serial.println("displayConnectionDetails");
  Serial.println("WiFi connected");
  Serial.println("IP address: ");
  Serial.println(WiFi.localIP().toString());
}

void startServer() {
  Serial.println("startServer");
  server.listen(80);
  server.on("/ip", [](PsychicRequest *request)
  {
    String output = "Your IP is: " + request->client()->remoteIP().toString();
    return request->reply(output.c_str());
  });
}
```

Produces the following compiler error:
```
PsychicCore.h:35:10: fatal error: esp_http_server.h: No such file or directory
 #include <esp_http_server.h>
          ^~~~~~~~~~~~~~~~~~~
compilation terminated.
exit status 1
```

hitecSmartHome, 2024-10-30:
Are you using arduino as a component of idf? If you are, enable http server in the menuconfig

hoeken, 2024-10-30:
I don't have any plans for that, but I would be happy to accept PRs if
people are interested and it's feasible.

hoeken, 2025-11-28:
Coming back to this after a long break and cleaning up issues.  Please re-open if its still a problem.