hoeken/PsychicHttp #198: After ~1 hour PsychicHttp can't establish new connections

kabell, 2024-11-23:
I'm running a webserver on esp32 using PsychicHttp.

It contains one webpage that has the following functionality:
- when it's loaded it opens a websocket connection. The server pushes some message every second to all connected clients using `websocketHandler.sendAll`. If the connection is lost, the webpage establishes a new websocket connection.
- every 5 seconds, the webpage hits one endpoint to fetch some data

Problem:
When I open the page, after ~1 hour, when I refresh the page, all the resources are loaded, but it's unable to establish a new websocket connection - it hangs.
When I want to open the page from another device, even the main page can't be loaded - it just hangs. At the same time I'm still able to load all the resources on the first device except the websocket.

At that state, the ESP has 150KB of free RAM, so it didn't go out of resources. My assumptions is, it somehow went out of connections, because new connections can't be opened, but the existing ones work perfectly.

So probably there is a bug in maintaining the connections, but I'm just guessing. This behaviour is 100% reproducable, and I can provide you any details if you tell me what and how.

Thanks for any help.

hoeken, 2024-11-24:
Are you on the main branch or v2?

kabell, 2024-11-24:
I use PsychicHttp v1.2.1

kabell, 2024-11-24:
I did another experiment - a simple page with a simple websocket(without any automatic refresh):
I opened the page on my PC (192.169.0.2) and on my phone (192.168.0.179).
Each of the devices opens 2 connections - one for websocket and one for other resources.

After I turn off the  phone display, in the log console I got
```
[ 65000][E][PsychivClient.cpp.62] remoteIP(): [psychic] Error getting client IP
```

After I turn the display  there were 5 connections - 3 from the phone. After I refreshed the page, there were 6 connections.
[image]
You can see on the image a log message with websocket clients count - active clients and server count - server clients
After each such log message there are several lines - each line contains local IP of the client and the remote IP of the client.

I did it again with th phone and I reached 7 connections and the ESP got not responsive.

hitecSmartHome, 2024-11-24:
https://github.com/hoeken/PsychicHttp/issues/184#issue-2541165992

hitecSmartHome, 2024-11-24:
Check lru_purge_enable and enable_so_linger options

hoeken, 2024-11-24:
Can you post a simplified sketch on GitHub with this issue?

kabell, 2024-11-25:
> Check lru_purge_enable and enable_so_linger options
Thanks a lot @hitecSmartHome 
`lru_purge_enable` helped, as it started recycling the connections.

closed 2024-11-25 by kabell