hoeken/PsychicHttp #204: Multiple definition of `httpd_req_async_handler_begin' and `httpd_req_async_handler_complete'.

tomikaka22, 2024-12-23:
Arduino IDE Version: 2.3.4 - Linux Appimage

I was using PsychicHttp on my ESP32C3 using the arduino IDE. The IDE offered to update my installed boards, and I'm not a 100% sure this is why, but now I can't even compile an empty sketch when I `#include <PsychicHttp.h>`.

```
/home/tomikaka/.arduino15/packages/esp32/tools/esp-rv32/2405/bin/../lib/gcc/riscv32-esp-elf/13.2.0/../../../../riscv32-esp-elf/bin/ld: /home/tomikaka/.arduino15/packages/esp32/tools/esp32-arduino-libs/idf-release_v5.3-083aad99-v2/esp32c3/lib/libesp_http_server.a(httpd_txrx.c.obj): in function `httpd_req_async_handler_begin':
/home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp_http_server/src/httpd_txrx.c:598: multiple definition of `httpd_req_async_handler_begin'; /home/tomikaka/.cache/arduino/sketches/9E68F6D620684A758A2CBFC10D8527F9/libraries/PsychicHttp/async_worker.cpp.o:/home/tomikaka/Arduino/libraries/PsychicHttp/src/async_worker.cpp:160: first defined here
/home/tomikaka/.arduino15/packages/esp32/tools/esp-rv32/2405/bin/../lib/gcc/riscv32-esp-elf/13.2.0/../../../../riscv32-esp-elf/bin/ld: /home/tomikaka/.arduino15/packages/esp32/tools/esp32-arduino-libs/idf-release_v5.3-083aad99-v2/esp32c3/lib/libesp_http_server.a(httpd_txrx.c.obj): in function `httpd_req_async_handler_complete':
/home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp_http_server/src/httpd_txrx.c:643: multiple definition of `httpd_req_async_handler_complete'; /home/tomikaka/.cache/arduino/sketches/9E68F6D620684A758A2CBFC10D8527F9/libraries/PsychicHttp/async_worker.cpp.o:/home/tomikaka/Arduino/libraries/PsychicHttp/src/async_worker.cpp:191: first defined here
collect2: error: ld returned 1 exit status

exit status 1

Compilation error: exit status 1
```

Example Code:
```
#include <PsychicHttp.h>

void setup() {
  // put your setup code here, to run once:

}

void loop() {
  // put your main code here, to run repeatedly:

}

```

Let me know any additional info I could provide.

tomikaka22, 2024-12-25:
Upon further testing it is the same on Windows and it seems that the latest update to the esp32 Arduino package is the culprit. Downgrading from 3.1.0 to 3.0.7 fixed the issue.

sfornengo, 2024-12-25:
Same problem for me

sfornengo, 2024-12-28:
Should this issue be declared on the esp32 arduino package (https://github.com/espressif/arduino-esp32 ) too ?

sfornengo, 2025-01-09:
Same with the last esp32 Arduino package version 3.1.1

sfornengo, 2025-01-13:
PsychicHttp functions httpd_req_async_handler_begin() and httpd_req_async_handler_complete() are already defined in expressif/esp-idf (file esp_http_server/src/httpd_txrx.c).
Removing PsychicHttp definition of this functions (in src/async_worker.cpp) seems to be the solution