hoeken/PsychicHttp #205: Ethernet support?

dzungpv, 2025-01-05:
Seem like the lib not support Ethernet. I try the lib with LAN 8720A chip but the Ethernet not working.

hitecSmartHome, 2025-01-05:
You have to put a compiler flag into pio.ini so the lib knows you are using ethernet.

hitecSmartHome, 2025-01-05:
**platformio.ini**
```cpp
build_flags = 
    -D PSY_ENABLE_ETHERNET=1
```

dzungpv, 2025-01-05:
> You have to put a compiler flag into pio.ini so the lib knows you are using ethernet.

You mean V2? I try to migrate but many error, seem not have any guide to migrate.
For example code in V1:

```
    // send to browser
    PsychicStreamResponse response(request, "text/plain");

    response.beginSend();
    response.print(html_common_header);
    response.print(optionPage);
    response.print(commonFooter);
    bool result = response.endSend();
```

Do you have commit or guide help migrate from the V1, I check this 5cbdc52 but it not all

hitecSmartHome, 2025-01-05:
Oh yeah, it's in V2 sorry. I belive v1 also supports ethernet. V2 is in development and there are no migration guide yet