hoeken/PsychicHttp #208: Client connections not closed when station is disconnected from soft AP

iduquegit, 2025-01-20:
When a station connects to the soft AP and I'm serving files (with multiple connections open), if the WiFi connection is abruptly closed, those connections remain open, and the `PsychicHttpServer::onClose()` callback is never triggered.

Should I manually close those connections when the station disconnects, or is there a better approach to handle this?

woodenplastic, 2025-01-20:
In V2 or main?

iduquegit, 2025-01-20:
> In V2 or main?

V2

hitecSmartHome, 2025-02-19:
I think this is somewhat related to my issue. I get a crash when a client vanishes and I want to send sse messages to it.
Psychic does not correctly closing some sockets in some cases and I think it wants to send packets to nullpointers.

## Crash

```cpp
Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.

Core  1 register dump:

PC      : 0x401ef11d  PS      : 0x00060b30  A0      : 0x801bcb41  A1      : 0x3fff4060

A2      : 0xc000e67e  A3      : 0x401bc8e4  A4      : 0x3ffd1444  A5      : 0x3f83f8ca

A6      : 0x0000000a  A7      : 0x0d7d5d7d  A8      : 0x00000000  A9      : 0x00000001

A10     : 0x00000000  A11     : 0x00000000  A12     : 0x00000018  A13     : 0x3ffeeaf0

A14     : 0x00000003  A15     : 0x000039cb  SAR     : 0x0000001e  EXCCAUSE: 0x0000001c

EXCVADDR: 0xc000e690  LBEG    : 0x4000c2e0  LEND    : 0x4000c2f6  LCOUNT  : 0x00000000

Backtrace: 0x401ef11a:0x3fff4060 0x401bcb3e:0x3fff4090 0x401bcb85:0x3fff40b0 0x401bce27:0x3fff40d0 0x401b12ad:0x3fff4100 0x401b16f7:0x3fff4140 0x400e34e8:0x3fff41f0 0x4016ab0d:0x3fff4260 0x4008afee:0x3fff4280

ELF file SHA256: 8604854bb

Rebooting...

```

## Decoded

```cpp

PC: 0x401ef11d
EXCVADDR: 0xc000e690

Decoding stack results
0x401ef11a: httpd_queue_work at C:\.platformio\packages\framework-espidf\components\esp_http_server\src\httpd_main.c line 147
0x401bcb3e: PsychicEventSourceClient::_sendEventAsync(void*, int, char const*, unsigned int) at .pio/libdeps/esp-wrover-kit/PsychicHttp/src/PsychicEventSource.cpp line 182
0x401bcb85: PsychicEventSourceClient::sendEvent(char const*) at .pio/libdeps/esp-wrover-kit/PsychicHttp/src/PsychicEventSource.cpp line 153
0x401bce27: PsychicEventSource::send(char const*, char const*, unsigned long, unsigned long) at .pio/libdeps/esp-wrover-kit/PsychicHttp/src/PsychicEventSource.cpp line 127
0x401b12ad: WebEventServer::finalSend(PsRamDoc&) at src/webServer/webEventServer/webEventServer.cpp line 99
0x401b16f7: WebEventServer::send(char const*, PsRamDoc&) at src/webServer/webEventServer/webEventServer.cpp line 45
0x400e34e8: std::_Function_handler   >::_M_invoke(const std::_Any_data &) at src/controllerHandler/controllerHandler.cpp line 375
0x4016ab0d: _FUN(void*) at C:/.platformio/packages/toolchain-xtensa-esp-elf/xtensa-esp-elf/include/c++/13.2.0/bits/std_function.h line 591
0x4008afee: vPortTaskWrapper at C:\.platformio\packages\framework-espidf\components\freertos\FreeRTOS-Kernel\portable\xtensa\port.c line 134

```

closed 2025-11-28 by hoeken