hoeken/PsychicHttp #212: void PsychicRequest::loadParams() method HTTP_DELETE and HTTP_PUT sent in body not being processed.

WeGoWireless, 2025-02-18:
PsychicRequst.cpp void PsychicRequest::loadParams() line 287 checks for HTTP_POST body _addParams but HTTP_DELETE and HTTP_PUT also send params in request->body(). I tried to add HTTP_DELETE and HTTP_PUT but may be failing due to contentType "application/x-www-form-urlencoded". Think it may be "text/plain". Thought I would post rather then attempting fix myself. More knowledgeable person may have better advice. 

Temp fix is calling my own addParams in my app.
addParams(request, request->body(), false);

void addParams(PsychicRequest* request, const String& params, bool post) {
	size_t start = 0;

	while (start < params.length()) {
		int end = params.indexOf('&', start);
		if (end < 0) end = params.length();
		int equal = params.indexOf('=', start);
		if (equal < 0 || equal > end) equal = end;
		String name = params.substring(start, equal);
		String value = equal + 1 < end ? params.substring(equal + 1, end) : String();
		request->addParam(name, value, true, post);
		start = end + 1;
	}
}

Thanks!

WeGoWireless, 2025-02-18:
One solution that worked for me is:
PsychicRequst.cpp void PsychicRequest::loadParams() line 287 

  if ((this->method() == HTTP_POST || this->method() == HTTP_PUT  || this->method() == HTTP_DELETE) && 
      (this->contentType().startsWith("application/x-www-form-urlencoded") || this->contentType().startsWith("text/plain")))

I don't think this covers every scenario like multipart/form-data. Would hope it doesn't break something else. Ideas welcome!

WeGoWireless, 2025-02-19:
Testing the below solution and so far working:
  if ((this->method() != HTTP_GET) && (this->contentType().startsWith("application/x-www-form-urlencoded") || this->contentType().startsWith("text/plain")))

hoeken, 2025-11-28:
Coming back to this after a long break and cleaning up issues. Please re-open if its still a problem.