hoeken/PsychicHttp #214: PsychicStaticFileHandler.cpp leaks open file handles

micethomas, 2025-02-20:
When PsychicStaticFileHandler.cpp sends the full file, it fails to close the file handle it uses to read the file content.
On or about line 174 I think you need to replace:

```
return response.send();
```
with something like this:

```
esp_err_t err = response.send();
_file.close();
return err;
```

WeGoWireless, 2025-02-20:
I am converting over to PsychHttp and just saw your message. I am an end user. Just thought I would comment. Looks like there is a 
PsychicFileResponse::~PsychicFileResponse()
{
  if(_content)
    _content.close();
}
on line 67 PsychicFileResponse.cpp. This is the file open "_content" so calling close on it is working for me. I just tested and it is getting hit. I downloaded about 20 files to test. Hope that helps.

micethomas, 2025-02-20:
> Oh my bad. I was looking at the PsychicFileResponse.cpp not the PsychicStaticFileHandler.cpp.

Thanks for taking a look though!