hoeken/PsychicHttp #216: Problems converting from ESPAsyncWebserver

designer-systems-ltd, 2025-03-13:
Hello

I am having some issues converting from ESPAsyncWebserver and what calls to use to send files on the file system using the request->reply().

Currently I setup the server for root as follows:

```
server->on("/", HTTP_GET, handle_root);
server->begin();
```
which handles the root file as follows:

```
void handle_root(AsyncWebServerRequest *request)
{
	// Redirect to index.html
	request->send(SPIFFS, "/index.html", "text/html");
}
```

For Psychic I have changed this to:

```
server.config.max_uri_handlers = 20;
server.listen(80);
server.on("/", HTTP_GET, [](PsychicRequest *request) {
	request->reply(200);
}, handle_root);

```
but I cannot find an example for sending a file using request->reply(), only a pre-configured file using the server.serveStatic(). As I would like to use this call to send differnet files and file types how can this be done ?

Regards David

zekageri, 2025-03-13:
What version of ESPAsnycWebserver are you migrating from? Because it just got a massive rework from @mathieucarbou and @me_no_dev and it is bettet than ever.

mathieucarbou, 2025-03-13:
> What version of ESPAsnycWebserver are you migrating from? Because it just got a massive rework from [@mathieucarbou](https://github.com/mathieucarbou) and @me_no_dev and it is better than ever. I recommend to not migrate to Psychic because it has some bugs and [@hoeken](https://github.com/hoeken) is pretty far away now to maintain this repo.

I know but he needs SSL ;-)

designer-systems-ltd, 2025-03-13:
>My understanding is that he needs SSL ;-)

 SSL is the ultimate goal, and ESPAsyncWebserver does not support that functionality, hence the need to move to Psychic.

>What version of ESPAsnycWebserver are you migrating from? Because it just got a massive rework from @mathieucarbou and @me_no_dev and it is better than ever. I recommend to not migrate to Psychic because it has some bugs and @hoeken is pretty far away now to maintain this repo.

We are using the latest repros of ESPAsyncWebserver, which is great for HTTP, but HTTPS still seems a long way off.

If anyone can help with the above it would be appreciated.

zekageri, 2025-03-13:
I have implemented https and I do not recommend it. It is slow and unreliable but yes I can help you with that. As soon as iam at my desk i will write the solution

designer-systems-ltd, 2025-03-17:
@zekageri started with just your example code but cannot get that to compile:

no instance of constructor "PsychicFileResponse::PsychicFileResponse" matches the argument listC/C++(289)

Does not like the 'res' object:

[image]

zekageri, 2025-03-17:
Are you on V2?

designer-systems-ltd, 2025-03-17:
V2 of what, latest version of PsychicWebserver is V1.2.1.

ESP packages:

PACKAGES:
 - framework-arduinoespressif32 @ 3.1.3
 - framework-arduinoespressif32-libs @ 5.3.0+sha.489d7a2b3a
 - tool-esptoolpy @ 4.8.6
 - tool-mklittlefs @ 3.2.0
 - tool-riscv32-esp-elf-gdb @ 14.2.0+20240403
 - tool-xtensa-esp-elf-gdb @ 14.2.0+20240403
 - toolchain-xtensa-esp-elf @ 13.2.0+20240530

hitecSmartHome, 2025-03-17:
Sorry. There is a V2 brach which is not published yet

zekageri, 2025-03-17:
Sorry, there is a V2 branch which is not published. https://github.com/hoeken/PsychicHttp/tree/v2-dev

designer-systems-ltd, 2025-03-21:
I have cloned the v2-dev branch using https://github.com/hoeken/PsychicHttp.git#v2-dev and everything is now compiling and I am able to see a very basic indelx.html page, thank you for the help.

Also, tried the SSL and although there are a number of errors being returned at the moment its definately somthing to builld on.

zekageri, 2025-03-21:
You can optimise it but it will be slow and the errors will not go away. :D i have tried optimising the hell out of it but eventually dropped the ssl thing

zekageri, 2025-03-21:
Glad it works hovewer!

designer-systems-ltd, 2025-03-21:
I have added in the other file parser ie. css, js, jpg, xml etc. and most of the page is working now apart from the CSS stylesheet.

I am getting a few of the following errors on each page load:

[xxxx][E][PsychicResponse.cpp:114] send(): [psychic] Send response failed (ESP_ERR_HTTPD_RESP_SEND)

My load_file() process is modified from the original ESPAsyncWebserver and looks like this:

```
bool load_file(PsychicRequest *request, PsychicResponse *response)
{
	String dataType = "text/plain";
	String path = request->url();

	// Process file type requested and set data type
	if(path.endsWith("/")) path += "index.htm";
	if(path.endsWith(".src")) path = path.substring(0, path.lastIndexOf("."));
	else if(path.endsWith(".html")) dataType = "text/html";
	else if(path.endsWith(".htm")) dataType = "text/html";
	else if(path.endsWith(".css")) dataType = "text/css";
	else if(path.endsWith(".js")) dataType = "application/javascript";
	else if(path.endsWith(".png")) dataType = "image/png";
	else if(path.endsWith(".gif")) dataType = "image/gif";
	else if(path.endsWith(".jpg")) dataType = "image/jpeg";
	else if(path.endsWith(".ico")) dataType = "image/x-icon";
	else if(path.endsWith(".xml")) dataType = "text/xml";
	else if(path.endsWith(".pdf")) dataType = "application/pdf";
	else if(path.endsWith(".zip")) dataType = "application/zip";

	// Open file and transfer to server
	PsychicFileResponse response_send(
		response,
		SPIFFS,
		path,
		dataType,
		false
	);
	response_send.send();
	return true;
}

```

So not sure why the style sheet is not being used as the files are loading:

[image]

designer-systems-ltd, 2025-03-21:
Loading the individual CSS files it looks like its the 'style.css' that is causing the error, I think because its being delivered to the page as 'chunked', whereas the other files just show a the content length.

b4iterdev, 2025-03-25:
I'm on an early development of something that are currently using ESP32AsyncWebServer and as I also need SSL (primary for websocket), does it work really well ? I read some comment above and still have no idea if they're working stable

designer-systems-ltd, 2025-03-25:
It works up to a point. If you are serving small HTML files then it will work but very slowly and with lots of errors.

zekageri, 2025-03-25:
Even with psram enabled, mbedtls needs internal ram for some things so the browser will repeat a couple of requests on each page load because the esp fails to allocate memory for the tls handshake process. Also it is slow to serve pages anyway, again because of the tls allocations.

zekageri, 2025-03-25:
And it is not implementation or library problem. It's just too much for the esp to use it in prod. Ram is limited in both space and speed. If you manage to use the psram for most things, it is even slower especially if you use the flash meanwhile. Psram and flash shares the same spi bus so they slow each other down.

zekageri, 2025-03-25:
In an internal network, even webservers written for a raspberry pi will use regular http.

b4iterdev, 2025-03-25:
what if the esp does not have to serve site but only handing secure websocket, with esp32webserver I can archive that with quite low latency

zekageri, 2025-03-25:
Wss is working. Iam doing that with the official esp websocket client implementation. As wss server, i dont know. In theory it would be more lightweight since it have to do the handshake only once on initial connection/client

b4iterdev, 2025-03-25:
I'm trying to get websocket up on my program:
```
// Global
PsychicHttpServer server;
PsychicWebSocketHandler websocketHandler;
// ...
void F1sh::initWebServer() {
    Serial.println("Webserver are being created");
    server.config.max_uri_handlers = 20;
    websocketHandler.onOpen([](PsychicWebSocketClient *client) {
      Serial.printf("[socket] connection #%u connected from %s\n", client->socket(), client->remoteIP().toString());
    });
    websocketHandler.onClose([](PsychicWebSocketClient *client) {
      Serial.printf("[socket] connection #%u closed from %s\n", client->socket(), client->remoteIP().toString());
    });
    websocketHandler.onFrame([](PsychicWebSocketRequest *request, httpd_ws_frame *frame) {
      Serial.printf("[socket] #%d sent: %s\n", request->client()->socket(), (char *)frame->payload);
      if(frame->type == HTTPD_WS_TYPE_TEXT) {
        Serial.printf("%s %s",(char *)frame->payload, frame->len);
      }
      return request->reply(frame);
  });
```
However, on init it's showing me this: ```[  2392][E][PsychicHttpServer.cpp:150] on(): [psychic] Add endpoint failed (ESP_ERR_INVALID_ARG)```
Does anyone know how to resolve this?

b4iterdev, 2025-03-25:
Also worth to note that trying to connect via websocat does not work

b4iterdev, 2025-03-25:
Also a dump question, in `websocketHandler.onFrame`, is there a way to not having return anything to the client at all ?

zekageri, 2025-03-25:
Well, idk. I think it requires you to return an esp_err type. Don't remember the cb. Will check it.

mathieucarbou, 2025-03-25:
@designer-systems-ltd : (little off-topic): by the way, what use case do you have that really required to setup an SSL server certificate on a MCU, which will highly probably be self-signed and not recognised by any browser, and which will then prevent anyone from browsing ?

If this is to protect creds, client web UI could use a MCU public key to encrypt them, then pass them to the MCU which will be able to decrypt them with its private key. Could be done as JWT token.

b4iterdev, 2025-03-26:
on my case specifically, I'm trying to develop a web application which is PWA that allow controlling robot with gamepad,... (a.k.a FtcRobotController but for ESP32). To make them "installable", it's required that the site must be served via https, which lead to websocket connection must also be secure.
atm I have to make a workaround that to install public cert into device and trust it, but as long as I only need to do it once, it's worth the trade-off.

zekageri, 2025-03-26:
> on my case specifically, I'm trying to develop a web application which is PWA that allow controlling robot with gamepad,... (a.k.a FtcRobotController but for ESP32). To make them "installable", it's [...]

I went to hell and back with this. You can't automate the cert install. I have tried a dozen different ways to make it understandable to the user and convinient but it's just not possible. Every possible platform and os implements it differently. It sucks. I also wanted to install PWA which the esp serves. It can work but really complicated for the user.

b4iterdev, 2025-03-26:
Yes it's a pity that there is no easy way to automate those, I even tried to find a way for websocket client to "skip cert validation" but apparently there is none

b4iterdev, 2025-03-26:
beside if you managed to use PWA, then I believe there's no point serving site via ESP32, just reduce huge load on esp and make the app as large as you want it to

zekageri, 2025-03-26:
Well, initially you have to serve the files somehow. I have tried to serve the pwa from a different origin ( from the cloud ) and i have tried the pwa to talk to the esp32 on the same network. I have managed to overcome cors issues but the self signed certs from the esp just made it impossible to use the pwa from a different origin.

b4iterdev, 2025-03-26:
AFAIK estabishing Websocket connection wouldn't mess with cors, this is my setup and it's completely working now.
https://github.com/F1sh-org/F1sh-quickstart/tree/wss_port - PlatformIO Project
https://github.com/F1sh-org/F1sh-static/tree/main - Angular Project

zekageri, 2025-03-26:
Well. To establish a wss connection you still need to do a cors request if the origin is not the same.

mhaberler, 2025-03-26:
@zekageri if this is just a POC, you could try a React Native Snack like this one: https://snack.expo.dev/@haberlerm/mqtt-client

the Expo Go binary on Android and iOS can open straight TCP/http/ws sessions

building a RN app isnt too hard, distribution is a bit of an uphill battle. Expo EAS helps.

b4iterdev, 2025-03-26:
> How do you load the app? Is it from the esp?

No, from another host: https://f1sh.b4iterdev.net.eu.org/

zekageri, 2025-03-26:
> > How do you load the app? Is it from the esp?
> 
> No, from another host: https://f1sh.b4iterdev.net.eu.org/

And you don't have CORS errors? O_o

zekageri, 2025-03-26:
> [@zekageri](https://github.com/zekageri) if this is just a POC, you could try a React Native Snack like this one: https://snack.expo.dev/@haberlerm/mqtt-client
>
> the Expo Go binary on Android and iOS [...]

Yeah, the most future proof and stable way would be to build an app which has no browser render engine but native and just do a connection to the esp32's IP or mdns name and we would not have to fight with CORS or certs :D

b4iterdev, 2025-03-26:
> > > How do you load the app? Is it from the esp?
> > 
> > 
> > No, from another host: https://f1sh.b4iterdev.net.eu.org/
> 
> And you don't have CORS errors? O_o

Yes, at first thought I have no idea how could you run into that issue, maybe checking Websocket client code, as I'm just simply `new Websocket(wss://192.168.4.1/ws)`

mhaberler, 2025-03-26:
> Yeah, the most future proof and stable way would be to build an app which has no browser render engine but native and just do a connection to the esp32's IP or mdns name and we would not have to [...]

that's where I landed with React Native Expo: https://github.com/mhaberler/mqtt-mdns/tree/master

resolve MQTT broker via mDNS, connect to service (MQTT/TCP in this case)

I gave up on MQTT/wss via this server
the cert management issue is unsolvable in an elegant way, and the primary dev is MIA again

zekageri, 2025-03-26:
> I mentioned earlier is an Android app and they comms using TCP)

Ohh so that is why you don't get CORS errors. You can even ignore any ssl issue altogether if you are writing an app. I tought it is a browser based web app and not a cordova / react native / expo / whatever app that has a native layer

zekageri, 2025-03-26:
> that's where I landed with React Native Expo: https://github.com/mhaberler/mqtt-mdns/tree/master

yeah, niiiceee

b4iterdev, 2025-03-26:
> Ohh so that is why you don't get CORS errors. You can even ignore any ssl issue altogether if you are writing an app. I tought it is a browser based web app and not a cordova / react native / expo / [...]

Thus the result is that I still can establish wss with no error as long as the cert is trusted by the system and the target is 192.168.4.1 as `subjectAltName` is set to on cert

b4iterdev, 2025-03-26:
> Yeah, this is still a very specific scenario. You are connecting to an ap mode which is an already isolated network. Also you manually install the cert to the system ( not automatically by the app code If I understand correctly ) and if the esp ip changes, the cert becames invalid

Yes, that's the flaw of it and project-specific. I have heard that you can obtain signed cert from creating a DNS record that somehow can point to the IP of ESP32