hoeken/PsychicHttp #219: Custom response code ignored in PsychicStreamResponse?

mirkosertic, 2025-06-05:
It seems setting a custom response code via `response->setCode(207)` is ignored in `PsychicStreamResponse`. It always returns a HTTP 200.

It fact, it seems there is no return code sent at all in the `beginSend? implementation of `PsychicStreamResponse`. The following implementation fixes this error:

```
  esp_err_t beginSend()
  {
    if (_buffer)
      return ESP_OK;

    // Buffer to hold ChunkPrinter and stream buffer. Using placement new will keep us at a single allocation.
    _buffer = (uint8_t *)malloc(STREAM_CHUNK_SIZE + sizeof(ChunkPrinter));

    if (!_buffer)
    {
      /* Respond with 500 Internal Server Error */
      httpd_resp_send_err(_request->request(), HTTPD_500_INTERNAL_SERVER_ERROR, "Unable to allocate memory.");
      return ESP_FAIL;
    }

    // THIS HERE SEEMS TO FIX THE PROBLEM
    //esp-idf makes you set the whole status.
    sprintf(_status, "%u %s", _code, http_status_reason(_code));
    httpd_resp_set_status(_request->request(), _status);
    // END OF FIX

    _printer = new (_buffer) ChunkPrinter(this, _buffer + sizeof(ChunkPrinter), STREAM_CHUNK_SIZE);

    sendHeaders();
    return ESP_OK;
  }
```

So the `httpd_resp_set_status` invocation is the missing part I guess, or maybe I am using this class wrong?

Chris--A, 2025-06-15:
Ah, yes, that is my mistake, `SendHeaders()` does not include the status code in `PsychicResponse`. So using the `stream.setCode(500);` will do nothing.

https://github.com/hoeken/PsychicHttp/blob/725f415a1352f86a3d47a94aee8ef9eec8aa2127/src/PsychicResponse.cpp#L119

@mirkosertic, what you have provided is the way `PsychicResponse` handles it in [`Send()`](https://github.com/hoeken/PsychicHttp/blob/725f415a1352f86a3d47a94aee8ef9eec8aa2127/src/PsychicResponse.cpp#L100) and would solve the problem. 

However, @hoeken, I feel a more appropriate place to send the status code would be in the [void PsychicResponse::sendHeaders()](https://github.com/hoeken/PsychicHttp/blob/725f415a1352f86a3d47a94aee8ef9eec8aa2127/src/PsychicResponse.cpp#L119) function, then all derived classes do not need to duplicate sending the status code (and it is a requirement to send status before additional headers).

mirkosertic, 2025-08-21:
Hi ! I don't want to be pushy here, but is there a change that a fix will be released in the near feature? I solved this issue by subclassing PsychicResponse and creating my own StreamResponse. However I'd like to use the core classes from the PsychicHttp package to reduce custom code on my side...