hoeken/PsychicHttp #224: PsychicHttpsServer no concurent connections?

dev-board-tech, 2025-08-20:
Try to use PsychicHttpsServer and connect from multiple browsers, seems that can't do it even  changing:
```
    serverHttps.ssl_config.httpd.max_open_sockets = 20;
    serverHttps.ssl_config.httpd.max_resp_headers = 20;
    serverHttps.ssl_config.httpd.backlog_conn = 20;
```

PsychicHttpServer seems to handle any incoming connection, everything I throw to it.

There is a catch on making it working with concurrent connections or concurrency is not implemented yet?

hitecSmartHome, 2025-08-21:
Unfortunately an esp32 does not have enough resources for concurrent ssl connections

hitecSmartHome, 2025-08-21:
Well, psram has a lot of limitations. Mbedtls cant really work with it for one. There are some buffers mbedtls can store in psram but not all. Also there are other hw limitations like you can't use the file system while writing or reading from psram because they use the same internal spi. We tinkered with ssl on an esp32 but it was barely usable even with one client. The whole self signed cert stuff around made us abandon ssl entirely... esp32 s3 and the p4 seems promising hovewer.

dev-board-tech, 2025-08-27:
s3 and p4 will be nice, s3 especially is more and more used being all in one chip.

closed 2025-08-27 by dev-board-tech