hoeken/PsychicHttp #226: Various compilation errors with the library

MicSG-dev, 2025-08-31:
The library gives several errors when trying to compile the project on PlatformIO.

- main.cpp:

```
#include <Arduino.h>
#include <PsychicHttp.h>

// put function declarations here:
int myFunction(int, int);

void setup() {
  // put your setup code here, to run once:
  int result = myFunction(2, 3);
}

void loop() {
  // put your main code here, to run repeatedly:
}

// put function definitions here:
int myFunction(int x, int y) {
  return x + y;
}
```

- platformio.ini:

```
; PlatformIO Project Configuration File
;
;   Build options: build flags, source filter
;   Upload options: custom upload port, speed and extra flags
;   Library options: dependencies, extra library storages
;   Advanced options: extra scripting
;
; Please visit documentation for the other options and examples
; https://docs.platformio.org/page/projectconf.html

[env:esp32doit-devkit-v1]
platform = espressif32
board = esp32doit-devkit-v1
framework = arduino
lib_deps = 
    hoeken/PsychicHttp
    bblanchon/ArduinoJson

```

- Compilation errors:

```
.pio/libdeps/esp32doit-devkit-v1/PsychicHttp/src/async_worker.cpp:130:32: error: 'HTTPD_MAX_REQ_HDR_LEN' was not declared in this scope; did you mean 'CONFIG_HTTPD_MAX_REQ_HDR_LEN'?
  130 | #define HTTPD_SCRATCH_BUF  MAX(HTTPD_MAX_REQ_HDR_LEN, HTTPD_MAX_URI_LEN)
      |                                ^~~~~~~~~~~~~~~~~~~~~
.pio/libdeps/esp32doit-devkit-v1/PsychicHttp/src/async_worker.cpp:127:22: note: in definition of macro 'MAX'
  127 | #define MAX(a, b) (((a) > (b)) ? (a) : (b))
      |                      ^
.pio/libdeps/esp32doit-devkit-v1/PsychicHttp/src/async_worker.cpp:138:29: note: in expansion of macro 'HTTPD_SCRATCH_BUF'
  138 |     char            scratch[HTTPD_SCRATCH_BUF + 1]; /*!< Temporary buffer for our operations (1 byte extra for null termination) */
      |                             ^~~~~~~~~~~~~~~~~
.pio/libdeps/esp32doit-devkit-v1/PsychicHttp/src/async_worker.cpp:130:55: error: 'HTTPD_MAX_URI_LEN' was not declared in this scope; did you mean 'CONFIG_HTTPD_MAX_URI_LEN'?
  130 | #define HTTPD_SCRATCH_BUF  MAX(HTTPD_MAX_REQ_HDR_LEN, HTTPD_MAX_URI_LEN)
      |                                                       ^~~~~~~~~~~~~~~~~
.pio/libdeps/esp32doit-devkit-v1/PsychicHttp/src/async_worker.cpp:127:28: note: in definition of macro 'MAX'
  127 | #define MAX(a, b) (((a) > (b)) ? (a) : (b))
      |                            ^
.pio/libdeps/esp32doit-devkit-v1/PsychicHttp/src/async_worker.cpp:138:29: note: in expansion of macro 'HTTPD_SCRATCH_BUF'
  138 |     char            scratch[HTTPD_SCRATCH_BUF + 1]; /*!< Temporary buffer for our operations (1 byte extra for null termination) */
      |                             ^~~~~~~~~~~~~~~~~
.pio/libdeps/esp32doit-devkit-v1/PsychicHttp/src/async_worker.cpp:130:32: error: 'HTTPD_MAX_REQ_HDR_LEN' was not declared in this scope; did you mean 'CONFIG_HTTPD_MAX_REQ_HDR_LEN'?
  130 | #define HTTPD_SCRATCH_BUF  MAX(HTTPD_MAX_REQ_HDR_LEN, HTTPD_MAX_URI_LEN)
      |                                ^~~~~~~~~~~~~~~~~~~~~
.pio/libdeps/esp32doit-devkit-v1/PsychicHttp/src/async_worker.cpp:127:35: note: in definition of macro 'MAX'
  127 | #define MAX(a, b) (((a) > (b)) ? (a) : (b))
      |                                   ^
.pio/libdeps/esp32doit-devkit-v1/PsychicHttp/src/async_worker.cpp:138:29: note: in expansion of macro 'HTTPD_SCRATCH_BUF'
  138 |     char            scratch[HTTPD_SCRATCH_BUF + 1]; /*!< Temporary buffer for our operations (1 byte extra for null termination) */
      |                             ^~~~~~~~~~~~~~~~~
.pio/libdeps/esp32doit-devkit-v1/PsychicHttp/src/async_worker.cpp:130:55: error: 'HTTPD_MAX_URI_LEN' was not declared in this scope; did you mean 'CONFIG_HTTPD_MAX_URI_LEN'?
  130 | #define HTTPD_SCRATCH_BUF  MAX(HTTPD_MAX_REQ_HDR_LEN, HTTPD_MAX_URI_LEN)
      |                                                       ^~~~~~~~~~~~~~~~~
.pio/libdeps/esp32doit-devkit-v1/PsychicHttp/src/async_worker.cpp:127:41: note: in definition of macro 'MAX'
  127 | #define MAX(a, b) (((a) > (b)) ? (a) : (b))
      |                                         ^
.pio/libdeps/esp32doit-devkit-v1/PsychicHttp/src/async_worker.cpp:138:29: note: in expansion of macro 'HTTPD_SCRATCH_BUF'
  138 |     char            scratch[HTTPD_SCRATCH_BUF + 1]; /*!< Temporary buffer for our operations (1 byte extra for null termination) */
      |                             ^~~~~~~~~~~~~~~~~
```

chrisfroe-roast, 2025-09-06:
Had the same. Current solution from my side: use `https://github.com/pioarduino/platform-espressif32/releases/download/53.03.13/platform-espressif32.zip`

MicSG-dev, 2025-09-08:
I found the solution to the problem: https://github.com/hoeken/PsychicHttp/pull/225/files

hoeken, 2025-11-28:
Merged and fixed

closed 2025-11-28 by hoeken